raspiblitz/raspiblitz #2777: Exeption on Subscription

ttrueten, 2021-12-08:
Hi there,
I got Permisson errors when I try to subscribe / generate a Let's encrypt certificate.

[image]

https://termbin.com/jf2m

Cheers,
Thomas

rootzoll, 2021-12-08:
Thanks for the report ... can you check if the following directory exists `/mnt/hdd/app-data/subscriptions` and if yes what are the permissions on that directory and also if exists the file `subscriptions.toml`with command `sudo ls -la /mnt/hdd/app-data/subscriptions/subscriptions.toml` on terminal?

rootzoll, 2021-12-08:
Note: Might be releated to issue #2689 already fixed for upcomming v1.7.2

ttrueten, 2021-12-13:
Hello,

drwxr-xr-x 2 bitcoin bitcoin 4096 Dec  3 10:33 subscriptions

the directory itself is empty.

rootzoll, 2021-12-13:
@ttrueten I think we found the reason and this should fix it:
https://github.com/rootzoll/raspiblitz/commit/236fc76e92a1fbb7bf2aa979180d6ad22d04fa39

Putting this issue on final testing for Release Candidates of v1.7.2

rootzoll, 2022-02-20:
OK tested again - works so far. CLosing for v1.7.2 release.

closed 2022-02-20 by rootzoll