raspiblitz/raspiblitz #2787: Hybrid clearnet and Tor mode for lightning (LND and CL)

openoms, 2021-12-10:
moving the discussion from #2684

## Hide the home IP address with a shared VPN

this should work on the RPi: https://mullvad.net/en/help/linux-openvpn-installation/

or a simple openvpn config and autostart with any provider (not recommending PIA): https://www.raspberrypi-spy.co.uk/2020/06/raspberry-pi-vpn-setup-guide/

To restart openvpn for the autostart to kick in:
```
sudo systemctl daemon-reload
sudo systemctl restart openvpn
```

## LND config flag
```
[tor]
; Allow outbound and inbound connections to be routed through Tor
; tor.active=true

; Allow the node to connect to non-onion services directly via clearnet. This
; allows the node operator to use direct connections to peers not running behind
; Tor, thus allowing lower latency and better connection stability.
; WARNING: This option will reveal the source IP address of the node, and should
; be used only if privacy is not a concern.
; tor.skip-proxy-for-clearnet-targets=true
```
## C-lightning setting
https://lightning.readthedocs.io/TOR.html#quick-start-on-linux
```
# 'always-use-proxy' informs C-Lightning to always use Tor even when connecting to nodes with public IPs. 
# You can set this to false or remove it, if you are not privacy-conscious and find Tor is too slow for you.
always-use-proxy=false
```

openoms, 2021-12-10:
Setting
` tor.skip-proxy-for-clearnet-targets=true`
for the signet LND instance makes it fail straight after the wallet unlock:

```
[ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
[INF] LNWL: Opened wallet
[INF] CHRE: Primary chain is set to: bitcoin
[INF] CHRE: Initializing bitcoind backed fee estimator in CONSERVATIVE mode
[INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on 127.0.0.1:23333
[INF] LNWL: Started listening for bitcoind block notifications via ZMQ on 127.0.0.1:23332
[ERR] RPCS: [/lnrpc.Lightning/GetInfo]: the RPC server is in the process of starting up, but not yet ready to accept calls
[INF] LNWL: The wallet has been unlocked without a time limit
[INF] CHRE: LightningWallet opened
[DBG] LNWL: Birthday block has already been verified: height=61982, hash=0000000c4c60c552336c6735f277fedc90f043bc8ab5b701660664722a8a69aa
[DBG] LNWL: Waiting for chain backend to sync to tip
[INF] LTND: Shutdown complete
````

openoms, 2021-12-14:
An initial script to activate a public VPN service + killswitch (firewall):

```
#!/bin/bash

sudo apt update
sudo apt install openvpn -y

## define config filename and credentials
#echo "# Type or paste the config filename and press ENTER:"
#read config; echo $config; echo
config="VPNconfig"
echo "# Type or paste the config filename and press ENTER:"
read username; echo $username, echo
echo "# Type or paste the config filename and press ENTER:"
read password; echo $password, echo

/home/admin/config.scripts/blitz.setconf.sh "/etc/openvpn/$config.conf" "root"
sudo chmod 600 /etc/openvpn/$config.conf

echo "\
$username
$password
" | sudo tee /etc/openvpn/auth.txt
sudo chmod 600 /etc/openvpn/auth.txt

sudo sed -i 's/auth-user-pass/auth-user-pass auth.txt/g' /etc/openvpn/auth.txt

time curl https://api.ipify.org
time curl https://api.ipify.org
time curl https://api.ipify.org
## Autostart
#sudo cp $your_config.ovpn autostart.conf
#sudo sed -i "s/#AUTOSTART=\"all\"/AUTOSTART=\"autostart\"/g" /etc/default/openvpn
#sudo systemctl daemon-reload
#sudo systemctl restart openvpn
#sudo mv /etc/openvpn/$config.ovpn /etc/openvpn/$config.conf

sudo systemctl start openvpn@$config

sleep 10
time curl https://api.ipify.org

## disable IPv6
echo "\
net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6=1
net.ipv6.conf.lo.disable_ipv6=1
" | sudo tee -a  /etc/sysctl.conf

sudo sysctl -p
if grep 1 < /proc/sys/net/ipv6/conf/all/disable_ipv6 ;then
  echo "# Successfully disabled IPv6"
fi
sudo sed -i "s/IPV6=yes/IPV6=no/g"  /etc/default/ufw
sudo ufw disable

## Killswitch
## https://www.comparitech.com/blog/vpn-privacy/how-to-make-a-vpn-kill-switch-in-linux-with-ufw/
## collect data
domain=$(cat /etc/openvpn/$config.conf | grep remote | head -n1 | awk '{print $2}')
echo $domain
port=$(cat /etc/openvpn/$config.conf | grep remote | head -n1 | awk '{print $3}')
echo $port
subnet=$(hostname -I | awk '{print $1}' | cut -d"." -f1-3)
echo $subnet

## UFW
sudo ufw allow in to $subnet.0/24
sudo ufw allow out to $subnet.0/24
sudo ufw default deny outgoing
sudo ufw default deny incoming
sudo ufw allow out on tun0 from any to any
for i in $(host $domain | awk '{print $4}' );do
    echo "# Allowing $i"
    sudo ufw allow out to $i port $port proto udp
done

sudo ufw --force enable

## test
sleep 10
time curl https://api.ipify.org
time curl https://api.ipify.org
sleep 10
time torsocks curl https://api.ipify.org
time torsocks curl https://api.ipify.org
```

openoms, 2021-12-15:
TODO for v1.7.2

remove the 
`setting ${lndConfFile} ${insertLine} "tor.streamisolation" "true"`
from lnd.check.sh
as it is incompatible with 
`tor.skip-proxy-for-clearnet-targets=true`
and will cause trouble when the lnd.conf is edited manually.

rootzoll, 2021-12-15:
OK looking foward to the PR :)

openoms, 2021-12-20:
For v1.7.2 `tor.skip-proxy-for-clearnet-targets=true` can be configured manually and `lnd.check.sh` will switch `tor.streamisolation` off on LND restart.

Pushing the better clearnet support to v1.8.

openoms, 2022-01-18:
A great guide about the options mentioned here and includes the option using a VPN: https://github.com/blckbx/lnd-hybrid-mode/

benjamin-wilson, 2022-03-09:
This is a killer feature, sooner the better! 🙏

openoms, 2022-03-09:
@benjamin-wilson if you are not worried exposing your public I address for the channel peers a simply setting:
`tor.skip-proxy-for-clearnet-targets=true`
in the `[tor]` section of the `lnd.conf` does the job.

In the v1.7.2 release `lnd.check.sh` already deals with (switches off) with `tor.streamisolation` if that option is active so shouldn't break anything.
I think the killswitch configs are out of scope of our implemenation, privacy focused users should keep tor-only.

So the implementation can be simply putting this setting in the menu with a sensible warning and have the similar solution (`always-use-proxy=false`) for C-lightning.

benjamin-wilson, 2022-03-10:
What is the behavior of LND when hybrid mode is on but the nat,externalip or externalhosts configurations are not set? Will it still skip tor for outgoing connections but not broadcast the public ip for incoming?

benjamin-wilson, 2022-03-10:
Will two nodes that both have `tor.skip-proxy-for-clearnet-targets=true` still communicate though tor only?

PrinzMonty, 2022-04-28:
That would be great! I currently tunnel traffic over OpenVPN, but I can't use Tor that way.

openoms, 2022-05-05:
@maximlomans to disable the tor streamisolation
edit it to false in the `config.scripts/lnd.check.sh`

since this is coming up repeatedly will disable this check completely for the next release

rootzoll, 2022-05-09:
@openoms lots of input here - can you give me the tl;dr as a take away for the v1.8.0 release .. sound ar last like enough interest for an FAQ entry on that topic.

openoms, 2022-07-04:
"Hybrid" mode settings for CLN:

Update` /home/bitcoin/.lightning/config:`

```
# Tor
addr=statictor:127.0.0.1:9051/torport=9735
proxy=127.0.0.1:9050
always-use-proxy=false

# Clearnet
bind-addr=0.0.0.0:<ClearnetPort>
announce-addr=<ClearnetIp>:<ClearnetPort>
```
See https://lightning.readthedocs.io/lightningd-config.5.html#networking-options

https://twitter.com/SwissRouting/status/1543617791017816064
https://t.me/lightningd/14227

openoms, 2022-07-09:
There is a great paid solution now under beta testing (looking very good): tunnelsats.com

turbolift, 2022-10-08:
i configured a ip2tor subscription for my raspiblitz to forward the public address to the onion service. after trying to use hybrid clearnet configuration i realized, that lnd.check.sh is "fixing" my manual configuration of lnd.conf.

is this a valid configuration and will raspiblitz support ip2tor configurations?

openoms, 2022-10-08:
> i configured a ip2tor subscription for my raspiblitz to forward the public address to the onion service. after trying to use hybrid clearnet configuration i realized, that lnd.check.sh is "fixing" [...]

No, this is not a good idea as the the outgoing connection will still go to your peers through your clearnet public IP address. You might as well broadcast your public IP as the node address, Ip2Tor doesn't help.
In any case you can just rename `lnd.check.sh` to stop it overriding your manual config.

benjamin-wilson, 2022-10-13:
I have hybrid mode working with LND but with CLN it's not connecting to any peers. Config:

```                                                                     
# Tor settings
proxy=127.0.0.1:9050
addr=statictor:127.0.0.1:9051/torport=9735
always-use-proxy=false

# Clearnet
bind-addr=0.0.0.0:9735
announce-addr=mydnsnamehere:42495
grpc-port=4772
```

logs:

```
2022-10-13T18:54:47.884Z INFO    plugin-backup.py: root        : Comparing backup version 620894 versus first write version 620895
2022-10-13T18:54:47.884Z INFO    plugin-backup.py: Comparing backup version 620894 versus first write version 620895
2022-10-13T18:54:47.884Z INFO    plugin-backup.py: root        : Versions match up
2022-10-13T18:54:47.884Z INFO    plugin-backup.py: Versions match up
2022-10-13T18:54:48.119Z INFO    connectd: Static Tor service onion address: "mytoraddresshere.onion:9736,127.0.0.1:9736" bound from extern port 9736 
2022-10-13T18:54:48.166Z INFO    plugin-bcli: bitcoin-cli initialized and connected to bitcoind.
2022-10-13T18:54:48.344Z INFO    lightningd: --------------------------------------------------
2022-10-13T18:54:48.344Z INFO    lightningd: Server started with public key 03db05f62fadeeeeb2e534afaf47522f86d2e4368c110fdc0a833e516937630a9b, alias 
```

Hangs there.

jchia, 2022-12-10:
> "Hybrid" mode settings for CLN:
>
> Update` /home/bitcoin/.lightning/config:`
>
> ```
> # Tor
> addr=statictor:127.0.0.1:9051/torport=9735
> proxy=127.0.0.1:9050
> always-use-proxy=false
>
> # Clearnet [...]

What is this file? It's not on my raspiblitz 1.8.0 and after I created it and restarted lnd.service, my node was still announcing only the onion address.

PrinzMonty, 2022-12-12:
> What is this file? It's not on my raspiblitz 1.8.0 and after I created it and restarted lnd.service, my node was still announcing only the onion address.

This will not work currently with this settings. Hybrid mode is on Milestone 1.8.2 https://github.com/rootzoll/raspiblitz/milestone/18

Currently, this is only possible through various workarounds!

One of them is to create a VPN server with a public static IP in front of your Raspiblitz. That's how I do it, but I don't recommend it because you have to change a few things and ensure that the settings are correct with every update. This may cause some of the features of your Raspiblitz to stop working properly. It is probably wiser to wait for the official integration ;)

benjamin-wilson, 2023-07-17:
Is there currently a way to do hybrid mode with CLN and 1.9? After 1.8 it broke the hybrid mode.