raspiblitz/raspiblitz #2791: Connect -> Reset function seems to break Thunderhub / RTL

P1xelfehler, 2021-12-12:
I wanted to renew the tls certificate using the CONNECT -> RESET function. After that I cannot login to Thunderhub or RTL giving me weird errors. Thunderhub tells me that I have to create an account first. And RTL wants me to unlock my wallet resulting in an endless loading spinner after typing in the password.

After running the CONNECT -> SYNC function, I can at least login to Thunderhub and RTL also seems to work.

But in Thunderhub the balances are at 0 and when trying to open the Channels page I get an ExpectedCommitmentTypeInChannelMessage error.

How to fix this? I have a backup of the lnd folder, can I fix it with that somehow?

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.1
chainnetwork: bitcoin / main
 20:03:23 up 19 min,  2 users,  load average: 0.78, 1.10, 0.99

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin-Daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-12-12 19:45:26 GMT; 17min ago
  Process: 3626 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 3628 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
 Main PID: 3683 (bitcoind)
    Tasks: 18 (limit: 4915)
   CGroup: /system.slice/bitcoind.service
           └─3683 /usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log

Dec 12 19:45:25 P1xlNode systemd[1]: Starting Bitcoin-Daemon...
Dec 12 19:45:26 P1xlNode systemd[1]: Started Bitcoin-Daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2021-12-12 19:44:06 GMT, end at Sun 2021-12-12 20:03:23 GMT. --
Dec 12 19:45:25 P1xlNode systemd[1]: Starting Bitcoin-Daemon...
Dec 12 19:45:26 P1xlNode systemd[1]: Started Bitcoin-Daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-12-12T19:46:20Z New outbound peer connected: version: 70016, blocks=713876, peer=5 (outbound-full-relay)
2021-12-12T19:46:27Z New outbound peer connected: version: 70016, blocks=713876, peer=7 (outbound-full-relay)
2021-12-12T19:46:30Z Socks5() connect to 42hmvh4cm4rh3nnfrma5t5kz5w2566setkvwy5iv4kkxqhiptws66hqd.onion:8333 failed: host unreachable
2021-12-12T19:46:36Z New outbound peer connected: version: 70016, blocks=713876, peer=8 (outbound-full-relay)
2021-12-12T19:46:45Z Socks5() connect to u2phcj2ccyv5aolmxq4obslzsn4mrkuytqhvab2ggkiy2uw7bwtesqyd.onion:8333 failed: host unreachable
2021-12-12T19:46:48Z New outbound peer connected: version: 70016, blocks=713876, peer=9 (outbound-full-relay)
2021-12-12T19:47:07Z New outbound peer connected: version: 70016, blocks=713876, peer=10 (outbound-full-relay)
2021-12-12T19:47:11Z New outbound peer connected: version: 70016, blocks=713876, peer=11 (outbound-full-relay)
2021-12-12T19:47:15Z New outbound peer connected: version: 70016, blocks=713876, peer=12 (outbound-full-relay)
2021-12-12T19:47:21Z New outbound peer connected: version: 70016, blocks=713876, peer=13 (outbound-full-relay)
2021-12-12T19:47:36Z Socks5() connect to a5wrswz7e4dt3t42dpdyap5irl3my5diu4dpy5x3obskb5vho4fbabyd.onion:8333 failed: host unreachable
2021-12-12T19:48:26Z Socks5() connect to oo4vxqeix7vltai6qa24nj2lnuc2nuqeweigtyk2dxeibye3k2hezfid.onion:8333 failed: host unreachable
2021-12-12T19:48:37Z New outbound peer connected: version: 70016, blocks=713876, peer=15 (block-relay-only)
2021-12-12T19:51:32Z New outbound peer connected: version: 70016, blocks=713876, peer=16 (block-relay-only)
2021-12-12T19:53:00Z UpdateTip: new best=000000000000000000086dbaf547ebb35901182b3282c4fe740fbe01cb431806 height=713877 version=0x2fffe004 log2_work=93.225608 tx=694406016 date='2021-12-12T19:52:33Z' progress=1.000000 cache=1.4MiB(10429txo)
2021-12-12T19:59:23Z Socks5() connect to cwsfn7p2tx3ihrtsksxjrivlbwgd5abjdse6pj3odkz4xduwazdjhaad.onion:8333 failed: host unreachable
2021-12-12T19:59:30Z New outbound peer connected: version: 70016, blocks=713877, peer=21 (outbound-full-relay)
2021-12-12T20:00:04Z Socks5() connect to mjxq5rmhkzpeowxedhltj6cjnvbep32rjnmgyq4eskhhdo7xo2yeteid.onion:8333 failed: host unreachable
2021-12-12T20:00:36Z Socks5() connect to dpnszvzrolaoddamtgqorahrkuoxq7n5hsjyosahg6r7cjcms3l3sjad.onion:8333 failed: host unreachable
2021-12-12T20:01:19Z Socks5() connect to z7ubmhh3lutmmxhig6qmaf3obhe2caigdx6q2ypnjw7xhou4xwweegid.onion:8333 failed: host unreachable

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - LND on
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-12-12 19:46:45 GMT; 16min ago
[... 292 lines cut ...]
bitcoinActive=1
blockchainPeers=12
blockchainHeight=713877
initialSync=0
syncProgress=100.00 
startcountLightning=3
lndActive=1
walletLocked=0
lndPeers=8
syncedToChain=1
scanTimestamp=1639338753
scanDate='Sun Dec 12 19:52:33 GMT 2021'
scanProgress=99.99
lndRPCReady=1
clActive=0
CLwalletLocked=0
blitzTUIActive=0
blitzTUIRestarts=0
vagrant=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
debug | torsocks nc termbin.com 9999

openoms, 2021-12-13:
Try  a restart (if the node wasn't restarted after the macaroons RESET) and reinstall the apps.

P1xelfehler, 2021-12-13:
I already did that. Also flashed the img again to the SD Card and loaded a backup. Still, in Thunderhub the balances are 0 and I get the ExpectedCommitmentTypeInChannelMessage error.

P1xelfehler, 2021-12-13:
Can I just copy the .macaroon files from an old backup into the current lnd folders (after stopping all processes of course)?

openoms, 2021-12-13:
you can do anything of you understand the scripts, but this best would be automated.
Did the apps reinstall work for you?

P1xelfehler, 2021-12-13:
nope

openoms, 2021-12-13:
so a full clean reinstall of thunderhub would be:
```
config.scripts/bonus.thunderhub.sh off
sudo rm -f /mnt/hdd/app-data/thunderhub

config.scripts/bonus.thunderhub.sh on
```
try it.

P1xelfehler, 2021-12-13:
Okay, I did this. There were no errors when running it, but unfortunately Thunderhub is still broken.

openoms, 2021-12-13:
not sure what the problem is, will try to reproduce.
You can also try updating thunderhub from the UPDATE menu

P1xelfehler, 2021-12-13:
An update of Thunderhub fixed it! 😱🥳
I cannot believe this.

openoms, 2021-12-13:
great that means this will be unlikely an issue in the next release

P1xelfehler, 2021-12-13:
thank you for helping 😊
Where can I send you some Sats?

rootzoll, 2021-12-13:
Can this issue be closed, then?

closed 2021-12-14 by openoms