raspiblitz/raspiblitz #2792: LNBits - wrong github reference

fjone, 2021-12-13:
I installed LNBits on my raspi but noticed a lot of new extensions didn't get installed.

It seems like LNBits development has moved from "https://github.com/arcbtc/lnbits" to this repo "https://github.com/lnbits/lnbits-legend"

https://github.com/rootzoll/raspiblitz/blob/a500cc7cf2727f9cf1c587fcca8307b72e9c4f66/home.admin/config.scripts/bonus.lnbits.sh#L264

openoms, 2021-12-13:
thanks for the report. Are all the extension get installed if you correct that line?

rootzoll, 2021-12-13:
lnbits repo was moved but the link should still point to the LNbits legend. The reason why an old version gets installed is because for raspiblitz by default the tag`raspiblitz` on the LNbits repo is installed. But @arcbtc promised to move the tag for the next release.

But it makes sense to change the link to the repo anyway - so marking that as TODO:
- [ ] adapt lnbits repo URL for v1.7.2
- [ ] make sure raspiblitz tag gets moved to new version

openoms, 2022-01-11:
LNbits seems to install ok, but don't know if should be concerned here
```
Installing collected packages: pylightning
Successfully installed pylightning-0.0.7.3
# processing assets
sudo: ./venv/bin/quart: command not found
# updating databases
sudo: ./venv/bin/quart: command not found

*** Updating Firewall ***
```
 after adding quart to the install:
` sudo -u lnbits ./venv/bin/pip install quart`

```
# processing assets
Usage: quart [OPTIONS] COMMAND [ARGS]...
Try 'quart --help' for help.

Error: No such command 'assets'.
# updating databases
Usage: quart [OPTIONS] COMMAND [ARGS]...
Try 'quart --help' for help.

Error: No such command 'migrate'.
```

rootzoll, 2022-01-13:
I think in the latest LNbits verison Quart got replaced with FastAPI so might be able to drop that extra install.

rootzoll, 2022-01-13:
Removed Quart - still installs :)

zanderisrael, 2022-01-14:
There is a new version with the lndpos extension, its not on the raspiblitz branch.
Is there a need for that tag at all? Why not install 0.4.0 that was released recently?

commit 88d77d41 (2021-12-16): fix #2792 & #2556 LNbits c-lightning (#2808)