raspiblitz/raspiblitz #2793: [RESEARCH] Letsencrypt Certificates for Local IP

rootzoll, 2021-12-13:
PROBLEM: If we provide a WebUI the user either would use it thru HTTP in the local network (which is unencrypted) or use a self-signed certificate which will display ugly Warnings in normal browsers and is a UX nightmare.

IDEA: There is a way to get a valid SSL certificate from lets encrypt for a local IPs ... but it would need a public dns server handing out subdomains for free (that could be spammed): https://esc.sh/blog/letsencrypt-ssl-for-local-domains/

So there could be a fulmo.org service handing out looooong random subdomains for everyone that wants and will return the the local IP of a raspiblitz (local IPs should not be a privacy concern).

The subdomain could even be a public-key ... and we create a random private key on the raspiblitz before setup for that service thats kept on the HDD/SSD later on so the user can make simple updates. Registration for a subdomain could be behind Tor onion service to protect users privacy (hide public IP).

The fulmo DNS server could still be a target to DDOS ... but RaspiBlitz could check first if the service is available (DNS resolves to the correct IP). If that service is not working the user defaults either to HTTP or a selfsigned cert.

To prevent use of the service for illegal stuff like black markets etc (liability risk for service provider) .. it would just accept local IPs and reject all others.

Of course that would be based on an open-source project (AGPL) to not introduce any lockin to Fulmo services.

TODO:
- [ ] research if similar services/projects already exist
- [ ] find somebody that would like to commit for that implementation as project leader

rootzoll, 2021-12-13:
@frennkie could you concept check that idea?

GoetzGoerisch, 2021-12-18:
For my local network I have setup most services like the following.

- Use FQDN for all services
- Letsencrypt certs with DNS challange
- internal DNS rewrites the FQDN IPs to the local net IPs

rootzoll, 2021-12-18:
@GoetzGoerisch interesting - how do you rewrite the internal DNS exactly - that would need manual changes on the laptop (like hosts file), right?

rootzoll, 2021-12-18:
> Ahhhh there might be a show-stopper: When the users browser (running on the laptop) makes a DNS resolve on that subdomain - it will reveal the users clearnet IP to the server, right?

After giving this some thought using DNS here should make the privacy situation a bit better then initally thought. Because when the user enters into the browser or is forwarded to the given Domain (from for example from the Fulmo server) the laptop will go to the first DNS resolver in his line .- which is the router. The router then will go to the DNS of the ISP or any other det DNS server in his config. Then that DNS server will do the final resolving to the Fulmo DNS server in our example that hosts the subdomain registry ... then the result is given down stream again until the laptop browser gets the local IP as a result. So the clearnet IP is NOT revealed to the FULMO server and that service cannot make a big survailance clearnetIP directory of all users using the service. BUT it is still not perfect, because the first DNS service in the line asked by the user at least now knows that this user is using such a service and is most prob runnigng a RaspiBlitz/node. So this is already better, but still not privacy perfect.

GoetzGoerisch, 2021-12-18:
> @GoetzGoerisch interesting - how do you rewrite the internal DNS exactly - that would need manual changes on the laptop (like hosts file), right?

This can be achieved by running a local dns resolver with filtering (pi-hole/adguard) which IP is distributed to all local clients by DHCP of the router. They provide a rewrite filter to deliver a local IP for a FQDN instead of the public IP.

GoetzGoerisch, 2021-12-19:
One additional note from my side, all subdomain names are available forever in the certificate transparency log.

https://tailscale.com/kb/1153/enabling-https/#machine-names-in-the-public-ledger