raspiblitz/raspiblitz #2797: (error) message(_provision.setup.sh fail) provision.setup.sh BECAUSE OF ERROR STATE (4)

kmanriqu, 2021-12-15:
Having trouble with initial setup. Migrating from MyNode. First attempt it detected old data and went through the migration setup. Had an issue with Lightning Progress ,kept rebooting. Reinstalled LND. Was not able to get in with new pw C saved nor raspiblitz. Ended up reset HDD keeping blockchain. First attempt similar PW issues. Setup pw and it would not recognize after reboot. 

Re-flashed sd and started again. Attempted to recover LND wallet from seed and backup file option. Then this log. Any input much appreciated. 

###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.7.1
Thu 28 Oct 11:00:58 BST 2021
***********************************************
# *** blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
No HDMI switch found. 
No FS EXPAND needed. needsExpansion(0) fsexpanded(1)
No SSHRESET switch found. 
No UASP FIX needed (1st-try).
Waiting for HDD/SSD ...
total 604
drwxr-xr-x   2 root root   4096 Oct 28 11:00 .
drwxr-xr-x 118 root root   4096 Oct 28 11:00 ..
-rw-r--r--   1 root root 565189 Jan 31  2020 moduli
-rw-r--r--   1 root root   1580 Jan 31  2020 ssh_config
-rw-r--r--   1 root root   3250 Jan 31  2020 sshd_config
-rw-------   1 root root   1381 Oct 28 11:00 ssh_host_dsa_key
-rw-r--r--   1 root root    606 Oct 28 11:00 ssh_host_dsa_key.pub
-rw-------   1 root root    513 Oct 28 11:00 ssh_host_ecdsa_key
-rw-r--r--   1 root root    178 Oct 28 11:00 ssh_host_ecdsa_key.pub
-rw-------   1 root root    411 Oct 28 11:00 ssh_host_ed25519_key
-rw-r--r--   1 root root     98 Oct 28 11:00 ssh_host_ed25519_key.pub
-rw-------   1 root root   1823 Oct 28 11:00 ssh_host_rsa_key
-rw-r--r--   1 root root    398 Oct 28 11:00 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    338 Mar  5  2021 ssh_import_id
wifiIsSet=0
hddBackupLocationAvailable=0
hddRestoreConfigAvailable=0
memRestoreConfigAvailable=0
wifiRestore=0
wifiBackup=0
No UASP FIX needed (2nd-try).
isMounted: 0
Skipping UASP deactivation ... cmdlineExists(1) hddAdapterUSB(174c:55aa) hddAdapterUSAP(1)
HDD already part of system: 0
Temp mounting (1) data drive (sda)
Temp mounting (1) result: 1
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
# df 
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       29574828   4380736  23945048  16% /
devtmpfs         1933976         0   1933976   0% /dev
tmpfs            1968600         0   1968600   0% /dev/shm
tmpfs            1968600     16828   1951772   1% /run
tmpfs               5120         0      5120   0% /run/lock
tmpfs            1968600         0   1968600   0% /sys/fs/cgroup
tmpfs              32768         4     32764   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095     31495    226601  13% /boot
tmpfs             393720         0    393720   0% /run/user/1000
/dev/sda1      960322016 427670056 483800536  47% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL 
NAME        FSTYPE LABEL
sda                
└─sda1      ext4   BLOCKCHAIN
mmcblk0            
├─mmcblk0p1 vfat   boot
└─mmcblk0p2 ext4   rootfs
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=0
hddRaspiVersion=''
isSSD=1
datadisk='sda'
datapartition='sda1'
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddBytes=1000141693440
hddGigaBytes=931
hddUsedInfo='408G (47%)'
hddDataFreeKB=483800536
hddAdapterUSB='174c:55aa'
hddAdapterUSAP=1

# RAID
isRaid=0
raidCandidates=0

# SWAP
isSwapExternal=0

Displaying Info Message: Please Login for Setup
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
# Sourcing /mnt/hdd/raspiblitz.conf 
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup 
# RASPIBLITZ SETUP STATE
network=bitcoin
hostname= Removed
lightning=lnd
network=bitcoin
chain=main
setPasswordA=1
setPasswordB=1
setPasswordC=1
seedWords='
seedPassword=''
staticchannelbackup='/home/bitcoin/temp/upload/channel.backup'
# df 
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       29574828 4283600  24042184  16% /
devtmpfs         1933976       0   1933976   0% /dev
tmpfs            1968600       0   1968600   0% /dev/shm
tmpfs            1968600   25040   1943560   2% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs            1968600       0   1968600   0% /sys/fs/cgroup
tmpfs              32768      16     32752   1% /var/cache/raspiblitz
/dev/mmcblk0p1    258095   31495    226601  13% /boot
tmpfs             393720       0    393720   0% /run/user/1000
tmpfs             393720       0    393720   0% /run/user/1001
# lsblk -o NAME,FSTYPE,LABEL 
# Update /home/admin/raspiblitz.info 
state=provision
message='Starting Provision'
baseimage=raspios_arm64
cpu=aarch64
board=rp4
ramMB=3844
ramGB=4
network=
chain=
localip=‘removed
online='1'
dnsworking=1
fsexpanded=1
displayClass=lcd
displayType=
setupStep=0
setupPhase='setup'
fundRecovery=0
hostname=
hddCandidate='sda'
hddGigaBytes=931
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddGotMigrationData=
# migrationFile()
# setting PASSWORD A
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (4)
This can also happen if _provision.setup.sh has syntax errors
# /home/admin/raspiblitz.provision-setup.log

###################################
# _provision.setup.sh
###################################
*** Prepare bitcoin ***
sudo: unable to initialize PAM: Too many open files
sudo: unable to initialize PAM: Too many open files
error='no data drive mounted'
File Exists: /mnt/hdd/bitcoin/bitcoin.conf --> 0
# setting PASSWORD B
# OK -> RPC Password B changed
# Reboot is needed
...
*** Start bitcoin (SETUP) ***
- This can take a while ..
FAIL: setup: failed bitcoin
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(_provision.setup.sh fail)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off

openoms, 2021-12-15:
To recover from seed + SCB you can use the v1.7 image as described here: https://github.com/rootzoll/raspiblitz/issues/2639#issuecomment-958021833

It will force close your channels.
If you have a latest copy of your LND directory or lnd-rescue-... file you can use that to recover keeping your channels.

openoms, 2021-12-20:
Did you get any progress here?

Since a lot has happened it is probably best to start a new node and recover from seed+SCB.

rootzoll, 2021-12-21:
> EXIT _provision.setup.sh BECAUSE OF ERROR STATE (4)

this is caused here:
https://github.com/rootzoll/raspiblitz/blob/6cd17ad27a901cc852fc734bd23500c02df02d57/home.admin/_provision.setup.sh#L125

Its exiting because the Btcoin-Testrun fails.

openoms, 2021-12-21:
@rootzoll would it help to dump the bitcoin debug.log and the journal for the bitcoin process here?

rootzoll, 2021-12-21:
Based on the different exit codes - different debug info would be needed. And that would then just be hidden in the `raspiblitz.log` - also for the coming webUI we need a way to front face the user. Basically this can be part of the new `blitz.error.sh` - but I would push this to v1.8

Ralwas, 2022-01-22:
Hallo Zusammen,

ich habe gestern meinen Raspiblitz erhalten und heute mit der Installation begonnen. Die Blockchain wurde schon sync.. Jetzt bin ich gemäß der hier für die Installation und Inbetriebnahme hinterlegte BEschreibung vorgegangen und bin bis zu dem Punkt gekommen, wo man die drei Passwörter vergibnt. Alle ausgewiesenen Schritte haben bis hier her funktioniert. Nach der 3 3-PW-Vergabe läuft der automatische Einrichtungsprozess, der aber Abgebrochen wurde.

Meldung:
OK - password A changed for user pi, root, admin & bitcoin

Calling_provision.setup.sh BEACAUSE OF ERROR STATE (4)
This can also happen if_provision.setup.sh has syntax errors
***************************************************************
ERROR - please report to development team
***************************************************************
state (error) message(Blockchain TEstrun)
https://github.com/rootzool/raspiblitz#support
command to shutdown --> off

Den abschließenden Befehl habe ich einegegeben und die Verbind zum Remote host wurde geschlossen.

Es ist die gleich Meldung die "openoms" erhalten hat. 

Wie ist jetzt die weitere Verfahrenweise. Bis zum Punkt ein Backup einzurichten kam ich noch nicht und einer Wiederherstellung ist es ja auch nicht.

Über eine Hilfestellung würde ich mich sehr freuen.

Beste Grüße

rootzoll, 2022-01-22:
@Ralwas Exit 4 bedeutet, dass der Bitcoin nicht started ... wenn du ne vorgesyncte Blockchaion erhalten hast, stimmt mit der ggf. etwas nicht (Fehler beim Kopierprozess). Daher am besten: Nochmal eine neue SD Karte flashen: https://github.com/rootzoll/raspiblitz#downloading-the-software und beim Start dann nicht "recover" sondern weitere Optionen und dort "Fresh Setup" wählen und dabei NICHT die bestehende Blockchain verwenden, sondern alles löschen

Ralwas, 2022-01-22:
Hey,

danke für die schnelle Rückmeldung. Das wäre ja ärgerlich wenn die vor
synchronisierte Blockchain nutzlos wäre, weil es ein Aufpreis war.

Ich probiere diesen Schritt aus, danke 🙏

radixrat, 2022-02-04:
Just ran into a similar issue on a migration to a new hard drive:
```
Calling _provision.setup.sh for basic setup tasks ..
EXIT _provision.setup.sh BECAUSE OF ERROR STATE (4)
This can also happen if _provision.setup.sh has syntax error
```

```
Process: 32004 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=1/FAILURE)
```

Noticed when I logged in as bitcoin user and tried to navigate to folders that I didn't have access to read past `/mnt/hdd` once I gave read perms to that folder the daemon restarted and started to work. May not be the same problem in every case but thought I would relay.

openoms, 2022-02-04:
@radixrat thank you for the report. In the dev branch (v1.7.2r1f currently) the disk folder permissions are taken care of, so hope this is fixed.

closed 2022-05-02 by openoms