raspiblitz/raspiblitz #2800: Sync Restarts Continuously

full-metal-bull, 2021-12-15:
[image]

Attempting to bind sparrow wallet to node IP address and entered the following commands below all the other settings in Bticoinconf with my IP, user and password (generic below). 

[image]

Once done the settings needed to be restarted to save. Now, sync it continues to restart the sync, and asking me that I continue to enter my password C over and over again.

openoms, 2021-12-15:
CTRL + C
type:
`menu`
see your bitcoinLOGS in SYSTEM
also
`sudo journalctl -fu bitcoind`

full-metal-bull, 2021-12-16:
Thank you, it solved my problem. Can you help with connecting to Sparrow Wallet? I am entering my ip as stated in their guide, but the connection has not been made. I can't figure out why.

openoms, 2021-12-16:
probably the firewall blocking.
Use the CONNECT menu bitcoinRPC option or Electrs.

full-metal-bull, 2021-12-16:
Y'all are the bomb! Thank you openoms

closed 2021-12-16 by full-metal-bull