raspiblitz/raspiblitz #2802: Different names of variables when datadisk is mounted or is not mounted

adevoss, 2021-12-15:
When executing `sudo /home/admin/config.scripts/blitz.datadrive.sh status`
different variables are listed in case the datadrive is mounted or not.

Other scripts (like _bootstrap.sh) use these variables as a source causing different behaviour than expected.

rootzoll, 2021-12-16:
Thats correct - but either the used vars are available in both situations or the bootstrap script should choose a different path if the hdd/ssd is mounted or not. Did you find a case where this goes wrong?

adevoss, 2021-12-16:
Yes I did. I'm testing https://github.com/rootzoll/raspiblitz/issues/2772 with a datadisk containing raspiblitz data.

When there is no `$hddCandidate` in the `_bootstrap.sh` script at some point, the script stops or expects a hdd present (can't remember exactly where/when).
When I rename two vars the script continues. Will create a pr.

adevoss, 2021-12-16:
In the second pr the names of the variables are added instead of renamed.

rootzoll, 2021-12-16:
Ah OK .. does this help with the VM setup case?

adevoss, 2021-12-16:
It helps, but all is not OK (yet) . See https://github.com/rootzoll/raspiblitz/issues/2772#issuecomment-996173266

rootzoll, 2021-12-16:
If it makes things easier - I will merge #2809