raspiblitz/raspiblitz #2810: [dev] [VM] Provisioning Thunderhub needs user interaction which is not possible

adevoss, 2021-12-16:
Updating Raspiblitz by using a fresh OS disk and
`sudo ./build_sdcard.sh false false rootzoll dev headless true false`
(see #2772)

When thunderhub is being provisioned this message appears in the log
```
Need to install the following packages:
next
Proceed ?
```
Can't input a 'Y'. Workaround for now:
`/home/admin/config.scripts/bonus.thunderhub on`

rootzoll, 2021-12-16:
This seemed to becaused because the new opt-out of the anon telemetry report of the next framework was before the install of the package. Its now after the install, but still before the build. In my test no manual user input as needed anymore.

adevoss, 2021-12-19:
Just tested dev. This was OK.

closed 2022-02-17 by rootzoll