raspiblitz/raspiblitz #2812: Can't unlock the C-Lightning Wallet

MinervaLND, 2021-12-17:
Hey guys! 
Now it is unfortunately it is my turn... 
My node is asking me to enter the password C for the mainnet C-Lightning wallet unlock. After entering the C password it says: Waiting to unlock wallet...

lnd-rescue file (160mb) and cl-rescue file (6.8 mb - after unpackaging empty) are already downloaded. 

Do you have any ideas how to solve this issue without recovering?

Many thanks in advance! 

[image]

[image]

openoms, 2021-12-17:
the manual command is:
`config.scripts/cl.hsmtool.sh unlock`

is there an error there?

also monitor CL for some unrelated error with:

`sudo journalctl -fu lightningd`

openoms, 2021-12-17:
Not sure why your CL rescue file would be empty if there is 6.8 MB something? Please look again

MinervaLND, 2021-12-17:
> the manual command is: `config.scripts/cl.hsmtool.sh unlock`
> 
> is there an error there?
> **There is no error just asking again to enter password C for mainnet c lightning wallet**
[image]

> also monitor CL for some unrelated error with:
> 
> `sudo journalctl -fu lightningd`

**Here I am getting this results**

[image]

MinervaLND, 2021-12-17:
> Not sure why your CL rescue file would be empty if there is 6.8 MB something? Please look again

CL rescue file just contains after unpackaging following folders:  mnt/hdd/app-data
all empty

openoms, 2021-12-17:
Will need to double check why this happened, but for now run:
`sudo rm /root/.cl.pw`
and unlock again

or switch the CL autounlock off

openoms, 2021-12-17:
Re. the rescue file it has a `.lightning` directory in app-data which is hidden by default.

MinervaLND, 2021-12-17:
> Will need to double check why this happened, but for now run: `sudo rm /root/.cl.pw` and unlock again
> 
> or switch the CL autounlock off

**After running    sudo rm /root/.cl.pw  I am getting the following message. How do I switch the CL autounlock off? **
[image]

MinervaLND, 2021-12-17:
> Re. the rescue file it has a `.lightning` directory in app-data which is hidden by default.

**Thanks 👍, hope we will figure out why this is happening...**

openoms, 2021-12-17:
> **After running    sudo rm /root/.cl.pw  I am getting the following message. How do I switch the CL autounlock off? **
> [image]
> 

press CTRL+C to enter the command line
type: `menu`
in CL - CLREPAIR

btw this should be resolved by restarting lightningd (or the whole node):
`sudo systemctl restart lightningd`

MinervaLND, 2021-12-17:
> press CTRL+C to enter the command line type: `menu` in CL - CLREPAIR
>
> btw this should be resolved by restarting lightningd (or the whole node): `sudo systemctl restart lightningd`

**In the CL Repair shall I now reset? **

MinervaLND, 2021-12-17:
Now the node is not asking anymore for the C passwort. 
[image]

openoms, 2021-12-17:
No, no need to reset, just switch the autounlock off if it is active.
If the restart does not solve it you can try to deactivate the encryption too.

openoms, 2021-12-17:
> Now the node is not asking anymore for the C passwort. [image]

Help a bit with describing what l you did.
Restart the node.
Look.at the logs as above and can post the link with the `debug` logs too. The command to get the link is the end after you have run:
`debug`

MinervaLND, 2021-12-17:
> > Now the node is not asking anymore for the C passwort. [image]
> 
> Help a bit by describing what you did. Restart the node. Look at the logs as above and can post the link with the `debug` logs too.

The command to get the link is the end after you have run: `debug`

https://termbin.com/5iuh

MinervaLND, 2021-12-17:
> > Now the node is not asking anymore for the C passwort. [image]
> 
> Help a bit by describing what you did. Restart the node. Look at the logs as above and can post the link with the `debug` logs too. The command to get the link is the end after you have run: `debug`

****

MinervaLND, 2021-12-17:
> > > Now the node is not asking anymore for the C passwort. [image]
> > 
> > 
> > Help a bit by describing what you did. Restart the node. Look at the logs as above and can post the link with the `debug` logs too.
> 
> The command to get the link is the end after you have run: `debug`
> 

I performed the degub command again:

https://termbin.com/jm11

openoms, 2021-12-17:
> I performed the degub command again:
> 
> https://termbin.com/jm11

As you can see the link is empty.
Try without the `torsocks` bit.

MinervaLND, 2021-12-18:
Here again without the torsocks bit, looks still quite empty...

https://termbin.com/evrk
https://termbin.com/87jv

openoms, 2021-12-18:
There might be some bigger issues with your node if the
`debug`
command does not work.
Consider getting a new SDcard and flashing the latest image.
C-Lighning, LND and other apps will.be recovered from the data on your disk.

MinervaLND, 2021-12-18:
The debug functions runs through... just the link to share with you does not work.

openoms, 2021-12-18:
copy and paste to hashb.in

MinervaLND, 2021-12-18:
https://hashb.in/#FQ18ZXTUFDAAQCUCCBlACgSQEIBlsAVALUQBEBRXAVQHFF8B5O9RBWTr7zuAIwA2ASwAuALwBuAUwBOAZyEB7AHYAuRAEYAdAHYtGuAGMAFgEMhy5VJEB3RTIDW6vqMOKLiAPSIAtueVwiAAMQaoArABsqgBMQYgArgAOiIhRACw6ADQpAMwJcrJy2YgCiqYAJoim0jKmAOZS6tppAJzZ2hFtmloAHDlwHEjolEQ0mJgAEhiUXoi4TExE6ERomIigcAVSJYp1cuqGpiJexoo+Up4VPhaetXKJQoKiYlqldZvWSZhmBQC0vwA+RDlFRSTYiI5Sf5AmRSCoATwGoDmqAAwgBpOjIJg0ABy5GGyAAathUTMNiJFL4VKJ7HNTIYHHUZIp4spKgUZBIhIZtocBAJ1HJ4iDEAArVkyZSmASGEQCRC/ABmiviiD4DKZLLZ5SRIDmzAxqKm2FxiAAFABZVCm3EjACUiHQAE1lpRLeQnURUKM2BtAPOk6tc7nZWk53N5irmwYsv3IpikPhUgRS+DK5Sk5XUpQqmYtnmshk8cnhchEifKxdL5Z8nhcIjcFnKYdkEakAG5EFJpYJM53pOy6YlYQUROpu6Ze+V7SnEKg5UJpOoGSJF9tzTI2coLHVHQplJGAGIyISIWLRDS/DTRK86TQ9VRBKI5O90S1ETsRYxVOqKQKW/xEBwch1ByCIcmiC160bdkZxSFIiFMOQHH2e8LWEa4x0QVoNDCOCUlROhtUSdQqzLRMw2EXk6xjUNwx5MF4KYpjABRSQAAUjAiCvHiOQZE8Up+Ro5QaIbENKl+coEyTZRFTcZQlQAXk8HxlBETxjHKStoJDESYK0OSVV+B5yiUlS1I0rTaN0sStGMxUMz4eI6jeJUhAEKRTNU9TNOsm4HKc15dgGcgpEMTQ7w0B8n1UF9EDuRI+FkGR4QeRAS3InxygAbQ0ABddR0AhGRV2UOpo1E2N40TFQtFquAQrCjQIqi5873ixKZGS1L0prbK8oKory0qXBaLjKSar1JB8AwIgDSYI0TTNK0bVxO0iEdShkGxZBGBYP0QE2EUqQleIpRlOUFV+NVtKbRU+EVX5lEUIz6lkRVlB6OB/kYXY5HVKQ6g8I5EGPU9z0va9b3vR80hiCJEDfIhsm7SpgfQYHwavG9IuCUJIhiOJEa0B76tC8Loei2L2qSlLTx6iscvyp1Bp3cqYLG6rlFqrRSca5rH1auKkISmnuurBn+uZ0xirzEaKuUDnpJ5wZGBmuaFpWi1rVtB0zziU1DyYXbWHWA7hVFCE3PevWvDM7zLPlzx/LqQL3kxyGmqIaIcjhmKWjIXEpBsRBWREPhWXZRBEikN65KsOVM3UGoFBUdQdBCDQImyQRFEZOQFJ0DQ0jCHosijmOZAUku73NHPGV+WEBFMeFfhUAR4Rnd3sZ0L2fYgmKejIGhEkk8siCEEjECsYPErLBSQgXxel6X6IEzhEIdHToIeiCFognKHRogiCoIkzlpyjSPh97SDRb76He0l5DQWiVSDjCkIQ6mMEQC6LkuekQMnJQyh54AA8chKkgVIBeOw6jRAAPp2EcApFoOQtDewyD0NIiARCgIUp0EuIQWgtAAaPDyAByLut5e4xWiKoDIJByFRxZMyKQch87aGXogQ4JgPLF3QZaIQuBzQ5ByD0PePRcGKE7kEC8WNqHeximEGKQQyDoFzihMI5pHRx1CscSkVQIhKlMAmMQigY4KDFOUAQMgIg2BkDkCQiQlSJAcMoNIYpjBCAcAlOQPhX7FWMGIE80REjKGMLUHIYQfB2AqFoGkqc+iiMQMYtyidECnDLAkZQsIGRmF7HAKhntFEvhihEAOQcQ7xDDhHSo0dY4qHjkNJOhRgFpwzlndUAkUK/2Lj0Ho2R6mVwiEEGuodw46mVPEAUDcpBNw7oU2RENu40LEY+HIQ8R6QnHpPae/057L0OYcsIQRTAXhaNEG+5zDARCkEqcROgpClyCHwVBORNI6D4P0wuT8IjlBCDkbB79P7f16SXFogDWkqDAbEEZC9sFvAQUghwKC0EYJ0C0O8uD8EtEIdEE5cQyEKUoUs+RxS+46HoRoRhzDdgjg4VoLhPD34KX4ToQRwjAXRD6GEKRMi5Eex7iUqKGy5jdJoKpNy6B4QHkzEQRQqJTqwlUqiMwFh1C4KbFIUBiAhB/UOA2LxpUEjJBUF0jRU9FCrlcga4BchFkCuxj0IgIQYZUoqcHcZtTy4NMsHo9JQDU6IC3pnbO3T86Fz6V0IZ+CwJQW6bM+ZrdlDt35csq8zrXW71ULfMgmArXdlXDKNKEJ3I4InsCaw/ryjZBsG5BUIhkr5FZtqxtpgqk1J1D6mQ6EkL6IrUkMhTQegRAAQUOS5Q/r1GkQ69NkUXWhGzU1D1HaJmRyGdwxp1aWnyDacGjpYaNERr/uIwZFd8E3wtF6yZSppkCETc3NNZLM2LpaDEaIK7r3rorpuv1CcsyQt3UGkNnS649MjeCs9sgL2QXNF+8oUyZmN0fbO59C6c2wzCNSp0R6tE6K3XKHBVIxBpAhAAR0THIJUDgbBwPqKAiQIgxRiDFCIeEYp4Q+DCFIRQ0QBDlAkIYHIq4D5ihyD4AQiUdBhEUHwYw0QdBiDkGR5sCS1BJLyKk9yAHMnHDZLkkwk53KoY9i+nNb6b55oLZK4tZYZTbFXMkDM5Z/21vrTgptPEW2gLbau71G7zRNyyY5k1Q6zytDHaFFQk6fwzqKWZ5+9CwifuqWuupP7dH/p3SnNQ+6gihvNXnMFp7u34LCGEK9qXamIfvchhZ8X0PRDoZETZZCdnqD2bPH+RyeshD4BoJ5GpjFhD4EqXk0DyjiNHZ0JU5QolpCPjoDIhgR0rdMC0Q4fQ0jw2BV/H+EG96AZyzChe0CggIt2Ei+wKLUHoMBT0DQcRsUEJ3pEaTwJITEoa66pr9CP1MOHHSthDKmV5L4XoMIHKRFhEPiOvlJmnWNYfJZxAw82sTw65Urr89etHLCKYc7hgdDvO2+UFol9uNX3eQyHIUgwhpCkBcpUxPz5KiMWkMnUgr75YyR/PbYKWgaCO8Ak7S8LtwMQdd1Fd20g9G9jgvBL3t7lcgkSkljqM1I5iIPAHLD6UKU4UvbhYPWU9C0GkKHYFi7p3h990IOQNC0Na9sjHU8sdsO67jo5PRxsREMIYMIL8Wh8B0BESIYQlQ5CCFbtIaRJxYefhBHQphR1pFfiXUdvOQX7b/hc4X0KgigNiATxesCrvINuxgvo8Nnu4p3uIw7au7cxUd9HmlgPWHsIN4yo3zK+Fm4t0IkRERttBF5aAuLpLTPodEUolLnbv2+qaQGqFuWQOHqKwd6IUHhnp3jRoh9Lc271an4j11Jz6H+0QIHT1VWu0bsy80/Pa+D2FfA7n7fpXU/7/rnV5NqaEdNdz9HcTlncx5XdOsPcccvcl5Bcgguc15UEwhDBJs4UIgms+AL4/d95adic+Ach8DIEF5RspBV4s9+cDs8hA0QFC8IEoEQhxdy8bs0V7tWgFccUS5b5Zc7wm9T8gC8ZQhi5289dgdu9QdeFWUWgtBIch8Xw0hzsIhbc+D51z9YYwI1FcNtFf0mkiMp5wklRFAIlDBH5FAHBZs+ApQwgcgBB2EwgJBQF4hDBEgyMHA+M+AbA0hwF4gIhEgBAchQFEpjBDASwIgHBTolR4RVNtxElRFNNzBtN1BdNskDN8ljMVZ8B8QtYVo1pHQXQ3QPQvQfQaB9pgA4BAwBA6JWwGIowMjPRkx4I0xcwAMcwMxKhzQCwGwyJeoujEx+JKiuQGJOwJwpx+wUYhwRxrBxwextN8I5wFwlwqh5ijhWZzRTBqlnoRxBpHRzRkA2FpksJtVRBfg3AMw9wLBIx0Zjh4tcZwhlEsMEZ3xOxogWgp1fwUxMAWReR2F1BYhys8hKBQFQpCppYRAPiPJ1IzgLgrgbgDJP4wxDATxEgRA5A+jmxeFGQwxvxhwPcQTfB/ArBjhzQTiPJDihpsg7MRAeJ55PB0AaBUQyR0B0B8IPjc5gcfinx7iASgTBolIeI+IBIZQhJUSHpYS6hXJ3IlJThzhfJhJXh2RUT9IVAVQiTFAMwFJSTMxySIRKSOFPBDwbR8AaBkBKA4IAIPBgJ2Sw8hcVS1SNSa0S0jgqSNA9SDSjSTTJpVZlhGBMjlodZ1pEBNptpjYSjDpRQTozpZR5RVQShI5fh7p/gnoXoGge1Hprwvpfgfo9h/pAYZJgZQY9YNcmoKZYYj4HikYuxI40YMZlCAFz8ohYgyziZ/heY0Jz9lFn4hZ7gOouo6ZxZMpGYBoQS8xaiQ4uYWycY2yCZOyRZOpaYYy+oegsMwgmY2QjMHMqRFBo4ZJR5JwkI2F1BVzexdDNzuwdh+RuEzA/V+M+ADhLyrAFRyh7pDhLArU4opAah9EqRhwlAe1qDxzay8ZlEGzqZZyxYMo+omYKjmx6JeR1AzSZJAcvi/o7TshiT1TQFRBNSHSdSDcXTsBDTjT/ybjyspyQKey0o+yILsx+i2x1BDx4i8w60RBvwRx9jEByFSTjjVSpByEeYGpWzALwg4gyK5z6Z+zJYoKWwBjYKnQqjIxdi7Nip0BQoFIo9kLQEHhYR7S5BeFyhplWZNiQS+KyYJzBKTlpzuzRLKKBz5ypLaKnRdLpk8xDLipxRZNsgXLjg3A2Rywe09UqhjgdB+ljLGoALbihKLLRZezwKbLCpNzo5KgRyJp+LTLwrzKRKwLepYqWZjUkquY6oUqwqSL7iMroqsrJZgSZZErMjkqTKiq7ihdSr5yspYUsMVzpj1yQ4tyPsIQNQChUJDyy0DETyZIBSFRDMrzHzbzTArzgQnyZqnpjhYQPzdDvy6Q/zCriKGrIrQKyqJZIKaKGI4LAJELgcuwMKyTN1bSLqsKKSnS8KCLKAiLJySrhZLLMr9rqLoK5LGgQYGLKgmKWK9iBBjgOKLquKMxeK4BM4chMUIh4bcUAytomBkB1ARzkBMBURdU/pFq3z7hosdxiYRy/B4QMlqhthAdewfA/pFKSoyokhiZVVQoHBYEihf1XIypxS2EqhI4xKOQfriYmAZA6hfhKBOp7B1AsoKjDBhBco4pEgwokoJarrtgFJUdpQJB4i1zK0dLEA1bH9gEuxxaZB1AMxda1aAAiNtZQfG4qdQMWlkHtGwLxMtcoIQGUVmN24tBsZIa8PQIIRlfQHNBeN9bQvRA4AjVcM1WEW9AocoC2j06aL0qPb0z0X01aXWA2I2ZgE2DYc2KkS2S6GSFO5SLyCyVE/iX6GU5SfE6wBUt4QA1Kki2ILQUdCrLKA2OW40cgXAdQJmxkVmAwntHS9+PwULSEbMSEMseBaghSJrbIR8me1fOetIRu+qgmVuxcxALKQMruiYHu9QXY2O1mAxZa2QY4JUFkHwOapeoDEBS5XQ0oGwN6WeprNerajetu7e3exgIgfEA8jq487qsai8ha [... cut ...] ByFOgt6pgDCh6ADYAJKAJL5RZyQdpuHbhPmLIBZymAEcDGAxKKXTmQPkHIBbhziGkilCHpGtAXMKNMKobAY1P8aN2mcGbhNQZuIHg6BQRGWYGATjNdjwIugQxDqkzFMRwqwUwMgDkAMGjMCpS2bLSR4MIZBKgZgXIJCoQgDQLsSd2yJGqISGHDJXr6gGUokEvS2vIQKcc2wHICQkMTBOh/QBiHUBnhsgI6Bvhe9gIB12DdhchN234a3aB4pQjf6ogcyAoBUk5CPDToIpQkkCfOHkJcgdG/QEkByANGiJF5SpQrTTWuIgI6GMg+RkHYB849tpHdI+RpgAVwHCKvRgY+RhMDYmmVqgh8h24EWgV2qnKfbx2hgInb64agT3jsAqkYk7OhL/vgjOmUFLpHEcMtP5FwA1eluFLOaYIyCB2xHCFH9sIgOFFccQoU8DiAowNgC6RBgIlFiAyUYq7FQ+cMRya0zIDNTdYmwJZpIkA7ucANhQAA==

openoms, 2021-12-18:
ah, maybe this is not the ideal pastebin 😃

anyway run:
`config.scripts/cl-plugin.backup.sh on`

(also your LND is unable to start because reverted to a previous version - you can use the UPDATE - LND menu to fix)

MinervaLND, 2021-12-18:
> ah, maybe this is not the ideal pastebin 😃
> **Yeah delet now the pastebin :) **

> anyway run: `config.scripts/cl-plugin.backup.sh on`
> 
**Got this result **
[image]

> (also your LND is unable to start because reverted to a previous version - you can use the UPDATE - LND menu to fix)

**right now updating the LND **

openoms, 2021-12-18:
So the solution was:

```
config.scripts/cl.hsmtool.sh autounlock-off
config.scripts/cl.hsmtool.sh unlock
```

Will need to check why there was a permission error with autounlock and it should fall back to asking the password instead.

openoms, 2021-12-18:
ok changes made in:
681e8c98 and 5afb6354 and cherry-picked to v1.7.

Until v1.7.2 the fixes are available in v1.7.1 when running `patch`.

rootzoll, 2022-02-16:
No further reports on this since RC2 .. closing issue.

closed 2022-02-16 by rootzoll