raspiblitz/raspiblitz #2813: [dev] lnd not running after install

adevoss, 2021-12-17:
After the final reboot lnd is not running:
`[WRN] LTND: Error validating config: ValidateConfig: error normalizing RPC listen addrs: parse address 0.0.0.0:10009 failed: dial tcp 127.0.0.1:9050: connect: connection refused`
(See #2772)

rootzoll, 2021-12-17:
I just tested with a fresh build of the sd card in `dev` brach and lnd is running on fresh install. 
Is your report from a scenario running a VM?

adevoss, 2021-12-19:
Yes. lnd still not started.
Where do I look for logs? `/mnt/hdd/lnd/logs/bitcoin/mainnet` only contain old *.gz files.

openoms, 2021-12-19:
you can use the alias `lndlogs` and the `SYSTEM` menu also
the lnd logs and path is in `debug` too

adevoss, 2021-12-20:
> use the alias `lndlogs`

Does not work.
`debug -l ` gives no link to share

`debug`:
`tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory`
...
```
*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-12-20T15:44:24Z connect() to *.*.*.*:9050 failed after wait: Connection refused (111)
2021-12-20T15:44:25Z connect() to *.*.*.*:9050 failed after wait: Connection refused (111)
```
...
```
*** LAST LND (MAINNET) ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Journal begins at Mon 2021-12-06 22:32:55 CET, ends at Mon 2021-12-20 16:44:34 CET. --
Dec 20 16:43:03 RASPIBLITZ-LIVE systemd[23479]: lnd.service: Failed to locate executable /usr/local/bin/lnd: No such file or directory
Dec 20 16:43:03 RASPIBLITZ-LIVE systemd[23479]: lnd.service: Failed at step EXEC spawning /usr/local/bin/lnd: No such file or directory
Dec 20 16:43:03 RASPIBLITZ-LIVE systemd[1]: lnd.service: Main process exited, code=exited, status=203/EXEC
```

openoms, 2021-12-20:
need find out why Tor does not work eg.:

`sudo journalctl -fu tor@default`

adevoss, 2021-12-20:
```
Dec 20 16:39:43 RASPIBLITZ-LIVE tor[908]: Dec 20 16:39:43.443 [warn] Error reading included configuration file or directory: "/mnt/hdd/app-data/tor/torrc.d".
Dec 20 16:39:43 RASPIBLITZ-LIVE tor[908]: Dec 20 16:39:43.443 [err] Reading config failed--see warnings above.
```
`ls -l /mnt/hdd/app-data/tor/torrc.d`:
```
-rw-r--r-- 1 root root  932 Dec 15 19:51 bridges
-rw-r--r-- 1 root root 1208 Dec 19 21:07 services
```

openoms, 2021-12-20:
I haven't come across this issue. The owner and permissions look fine.
Have a look what is inside those files if there is anything weird.

adevoss, 2021-12-20:
> Is your report from a scenario running a VM?

Missing or moved files not only apply to a VM setup I think.
Are files moved recently and not match anymore with unit file of lnd service?

adevoss, 2021-12-22:
See https://github.com/rootzoll/raspiblitz/issues/2772#issuecomment-999675050

adevoss, 2021-12-23:
In `build_sdcard.sh` lnd is only installed when the option 'fatpack' is true. It is false by default so lnd is not installed most of the time. Don't know if this is by design or not.

Workaround: `sudo lnd.install.sh on` after installation is finished.

adevoss, 2021-12-23:
After #2846 and #2847 the lnd service is still not started.
Can't see why.

adevoss, 2022-01-02:
Tested the `dev` branch with this workaround after the setup: `sudo lnd.install.sh on mainnet` after installation is finished.
Result:
```
sudo tail -f /home/bitcoin/.lnd/logs/bitcoin/mainnet/lnd.log
2022-01-02 23:10:09.226 [WRN] LTND: Error validating config: ValidateConfig: error normalizing RPC listen addrs: parse address 0.0.0.0:10009 failed: dial tcp 127.0.0.1:9050: connect: connection refused
2022-01-02 23:11:10.437 [WRN] LTND: Error validating config: ValidateConfig: error normalizing RPC listen addrs: parse address 0.0.0.0:10009 failed: dial tcp 127.0.0.1:9050: connect: connection refused
2022-01-02 23:12:10.972 [WRN] LTND: Error validating config: ValidateConfig: error normalizing RPC listen addrs: parse address 0.0.0.0:10009 failed: dial tcp 127.0.0.1:9050: connect: connection refused

sudo journalctl -fu lnd
sudo /usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf
same as above
```

adevoss, 2022-01-13:
Tested sudo ./build_sdcard.sh --interaction true --fatpack true --github-user rootzoll --branch dev --display headless --tweak-boot-drive false --wifi-region off
Now syncing (see #2772)

closed 2022-02-17 by rootzoll