raspiblitz/raspiblitz #2820: ssh keys backup refactor

rootzoll, 2021-12-18:
see #2818 & # #2819 - merged for final testing

TODO: create a v1.7.1 Blitz - then insert a RC1 sd card and check if same SSHd key (no changes keys warning) and backup and conversion worked correctly (already before starting setup)

openoms, 2021-12-19:
having an issue with the ssh connection during a failed setup:
if the disk fails to mount (due to hardware issues) the raspiblitz restarts automatically and the ssh service fails to start afterwards.

openoms, 2021-12-19:
thanks, the above is fixed in e66848a

openoms, 2021-12-26:
Had a tough time on Ambian (Odroid HC2) after recovering my old disk:
```
Permission denied (publickey).
```

It did not get solved when placed the `ssh.reset` file in the `/boot/` dir on the sdcard.

The only way to get in was to create a file on the SDcard at `/home/admin/.ssh/authorized.keys` and paste my laptop's ssh pubkey to it. 
It was still asking for password A so there are some quite tough default values on the ssh server.
Would like to find out how to improve this, but until then this is a note for other Armbian users.

rootzoll, 2022-01-16:
@openoms If I check on the sdcard image rc1d with bullseye the `/etc/ssh/sshd_config` there is no entry for `AuthenticationMethod`

Should we set it explicitly? Or what are the measures to take here?

rootzoll, 2022-01-26:
@openoms this is one of the last things to address before building a new RC - whats the best direction here from your perspective?

openoms, 2022-01-26:
I think the problem is that if key-auth only was set the setting file is restored, but the `.ssh/authorized_keys` directory is not.

Should use `cp -ra` instead of just `cp -a` in the backup and restore function.

closed 2022-02-17 by rootzoll