raspiblitz/raspiblitz #2821: VPS & Tor

Tanki90, 2021-12-18:
Hello, 

I configured my Raspiblitz in Clearnet with a public ip from DigitalOcean
Tutorial: https://www.mobycrypt.com/turn-your-self-hosted-lightning-network-node-to-public-in-10-minutes/ 

Is it possible to run the node in Clearnet for requests etc. and activate Tor only for Applications like LNBits, Zeus Wallet, Sphinx Chat to have better security?

openoms, 2021-12-18:
It is possible, but corrently not implemented. Tor only is default and you could manually activate the clearnet access for LND.
DynDNS does not provide any security or privacy since it is just resolves to the given IP address.

See: https://github.com/rootzoll/raspiblitz/issues/2787

Tanki90, 2021-12-18:
Yeah, but I deactivate ''run behind tor'' in the menu and configure my node with the tutorial above to a public ip...
To get access with applications like lnbits, zeus..... I need DynDNS and open ports in the VPS droplet (those are my security concerns)

My thoughts are to connect only the Applications over a Tor Adresse.
Maybe with Hybrid Mode....idk

Tanki90, 2021-12-18:
aah ok thx. you are the man

closed 2023-06-07 by cstenglein