raspiblitz/raspiblitz #2824: host not found in upstream "mempool.space

rootzoll, 2021-12-19:
With updated nginx (based on bullseye repo) the nginx conf of mempool is throwing an error:

```
Dec 19 21:35:19 raspberrypi nginx[703]: nginx: [emerg] host not found in upstream "mempool.space" in /etc/nginx/snippets/mempool.conf:38
Dec 19 21:35:19 raspberrypi nginx[703]: nginx: configuration file /etc/nginx/nginx.conf test failed
Dec 19 21:35:19 raspberrypi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Dec 19 21:35:19 raspberrypi systemd[1]: nginx.service: Failed with result 'exit-code'.
Dec 19 21:35:19 raspberrypi systemd[1]: Failed to start A high performance web server and a reverse proxy server.
--> CHECK CONFIG: sudo nginx -t
```

Trying to remove this lines for now ...

```
	location /api/v1/donations {
		proxy_pass https://mempool.space;
	}
	location /api/v1/donations/images {
		proxy_pass https://mempool.space;
	}
```

rootzoll, 2021-12-19:
@Emzy @wiz dont wanna prevent donations here ... but if you like to keep a redirect to the mempool.space donations site/api here, please provide a PR with a working nginx config.

wiz, 2021-12-20:
This is likely due to your Internet connection not being available right away. We could hard code an IP address as a workaround, or maybe nginx has a more clean fix. Will get back to you shortly

wiz, 2021-12-20:
@rootzoll it looks like stackoverflow has a simple fix for this, is your systemd script missing the `network-online.target` ? https://serverfault.com/questions/749662/nginx-fails-to-start-on-boot-with-systemd-on-ubuntu-15-04-and-15-10

wiz, 2021-12-20:
what's likely happening is that the user powers on their router and raspberry pi at the same time, and the raspberry pi boots up faster than the router gets an external internet connection online, so for the first few minutes the raspberry pi is on it has a LAN connection, but no internet connection, so it cannot resolve mempool.space, and fails to start nginx since it doesn't wait for that proper "network online" event

openoms, 2021-12-20:
thanks for the analysis @wiz.
This hopefully will be solved as just added
```
After=network-online.target
Wants=network-online.target
```
to the bitcoind.service in https://github.com/rootzoll/raspiblitz/commit/4636f47aad450287b9054bb8287c421fc6795f2f

on which mempool is dependent on.

Will test in the next RC before v1.7.2.

rootzoll, 2021-12-20:
Thanks - sure we can give it a test run for v1.7.2

rootzoll, 2021-12-20:
OK its back in and set for final testing on v1.7.2

wiz, 2021-12-20:
Did you see the resolver option? There might be an even better way to set nginx to resolve it on demand using the 127.0.0.1 resolver to avoid having it fail to start if mempool.space cannot be resolved for some reason.

openoms, 2021-12-20:
On demand would be much better since people will wonder why we are pinging mempool.space on clearnet.

wiz, 2021-12-24:
> people will wonder why we are pinging mempool.space on clearnet.

To be clear, this is not a "phone home" or anything like that, it's an nginx.conf proxy so if you hit the donations/contributors API via the About page it will connect to us, but you made a good point that this should be done over Tor - we can improve that

wiz, 2021-12-28:
@rootzoll I confirmed adding "resolver 8.8.8.8;" will further allow nginx to start even if internet is unavailable, I suggest you also add this as part of the fix for this issue

rootzoll, 2022-01-04:
Thanks ... going with resolver 1.1.1.1 (cloudflare) here instead of 8.8.8.8 (google) .. possible potädo potado, but lets see ;)
https://blog.cloudflare.com/dns-resolver-1-1-1-1/

TODO: final testing