raspiblitz/raspiblitz #2825: electrs after `update/recover` missing bitcoin RPC data in config

rootzoll, 2021-12-19:
After a recover/update run:
`sudo nano /home/electrs/.electrs/config.toml`
-->
```
timestamp = true
jsonrpc_import = true
index-batch-size = 10
wait_duration_secs = 10
jsonrpc_timeout_secs = 15
db_dir = "/mnt/hdd/app-storage/electrs/db"
auth = ":"
# allow BTC-RPC-explorer show tx-s for addresses with a history of more than 100
txid_limit = 1000
server_banner = "Welcome to electrs v0.9.3 - the Electrum Rust Server on your RaspiBlitz"
```

So in `auth` there is missing:
https://github.com/rootzoll/raspiblitz/blob/cf8d45498877c5f8ba3045b2e6d48a0710abde17/home.admin/config.scripts/bonus.electrs.sh#L312

rootzoll, 2021-12-20:
hmmm - maybe its just a some quirk, but to maybe keep an eye on -  on testing a reinstall of electrs its haning forever on installing at this point:
[image]

rootzoll, 2021-12-20:
OK about that report above "it went thru" after all ... but really took a lot of time.

rootzoll, 2021-12-20:
When re-installing after recovery/update the `auth` value looks fine (already with that little fix). But it may be related about the install during recovery. To be rechecked.

rootzoll, 2022-01-17:
OK found the problem  ... on BTRFS the HDD was premounted by new background.scan process and so had problems to read out the bitcoin.conf password B to set electrs config correctly. Thats now fixed and test worked. Closing issue.

closed 2022-01-17 by rootzoll

openoms, 2022-01-17:
Building electrs takes a lot of time. Would be nice to get binary either we could brebuild and ship with the image or use the cryptoanarchy repo maybe?

rootzoll, 2022-01-18:
We could build as part of the fatpack during sd card build in the future?