raspiblitz/raspiblitz #2827: Sample lnd.service file

openoms, 2021-12-20:
There should be some changes in our default lnd.service file according to:
https://github.com/lightningnetwork/lnd/blob/master/contrib/init/lnd.service

openoms, 2021-12-21:
Testing wallet reset and LND does not want to shut down and hangs here:
```
Starting Lightning Network Daemon on mainnet...
Stopping Lightning Network Daemon on mainnet...
[lncli] rpc error: code = Unknown desc = wallet recovery in progress, cannot shut down, please wait until rescan finishes
lnd.service: Control process exited, code=exited, status=1/FAILURE
```
will need to return to the less graceful option.

rootzoll, 2022-02-17:
ok lnd install working fine in RC2 ... closing this issue for release.

closed 2022-02-17 by rootzoll