raspiblitz/raspiblitz #2828: [1.7.2RC1]

openoms, 2021-12-20:
```
# OK - the chain is synced
# Calling:
sudo -u bitcoin /usr/local/bin/lncli -n=mainnet --rpcserver localhost:10009 newaddress p2wkh

{
    "address": "xxxxxxxxxxxxxxxxxxxxxxxxx"
}
generating QR code ... please wait
/home/admin/BBfundWallet.sh: line 85: ln_${LNTYPE}_${$chain}net_wallet_onchain_pending: bad substitution
/home/admin/BBfundWallet.sh: line 86: ln_${LNTYPE}_${$chain}net_wallet_onchain_balance: bad substitution
```

`eval` or similar needs to be used in:
https://github.com/rootzoll/raspiblitz/blob/c3a4eb3bfbdfd9769c8d796b1523fb844da1acd0/home.admin/BBfundWallet.sh#L85

adevoss, 2021-12-20:
There is a double '$' sign in this line. Removing the second one will solve it I think.

openoms, 2021-12-20:
yes, that should be it, thanks @adevoss.

openoms, 2021-12-20:
there has been a few more so will leave this open for final testing

rootzoll, 2022-02-16:
Hope we got them all .. no further reports since RC2 release. Closing issue.

closed 2022-02-16 by rootzoll