raspiblitz/raspiblitz #2832: add REPAIR-LND menu to offer restore from file, seed+SCB and seed only

openoms, 2021-12-20:
Currently testing

rootzoll, 2022-01-04:
OK the PR changed quite a lot because if the change to the new proto files. Because a test with a new sd card build is needed I merge it and then now it needs to be final tested on the following details:

- [ ] new sd card build with merged PR - running?
- [ ] recover with SEED for on-chain funds
- [ ] recover with SEED & SCB for on-chain funds & closing channels
- [ ] run/test Blitz TUI
- [ ] run/test Wallet-Auto-Unlock (or change to new LND method)

openoms, 2022-01-16:
The route - REPAIR- LND - SEED+SCB has a problem - investigating
[image]

rootzoll, 2022-01-18:
Needed to add the "scan window" again to `lnd unlock` & `iniwallet from seed` to make lnd find the on-chain funds again.

openoms, 2022-01-18:
> Needed to add the "scan recovery window" again to `lnd unlock` & `iniwallet from seed` to make lnd find the on-chain funds again. 

It is there for the restore from seed: https://github.com/rootzoll/raspiblitz/blob/2ded9deb981cb0d374473dea7b098a185a8a6533/home.admin/99lndRepairMenu.sh#L440

and for seed+SCB: https://github.com/rootzoll/raspiblitz/blob/2ded9deb981cb0d374473dea7b098a185a8a6533/home.admin/99lndRepairMenu.sh#L269

rootzoll, 2022-01-18:
On setup the actual scan happens after the reboot and thats were the normal unlock is used

openoms, 2022-01-18:
That is right. The rescan is done if the REPAIR-LND menu is used (without restart).
If restoring with a fresh setup the rescan would need to be done manually from the menu after the initial setup and bitcoind sync.

If this is documented clearly might be enough or at least where I got to.

(This is also an issue for CL - the rescan there is fully manual, only documented in the FAQ.cl.md - could add some new logic there too)

DocBrown101, 2022-01-19:
I have made one step forward but now I get a new error message:
```
WALLET --> UNLOCK WALLET - SCAN 0
WALLET --> SEED + SCB
lnd-wallet-seed+scb lnd.initwallet.py scb returned error /home/admin/config.scripts/lnd.initwallet.py scb mainnet ... --> the given filepathSCB - file does not exist or no permission +

Press ENTER to return to main menu.
```

rootzoll, 2022-01-24:
The repair options got soem attention with this issue #2901 @DocBrown101 if you have some time best to wait for v1.7.2

closed 2022-02-20 by rootzoll