raspiblitz/raspiblitz #2836: eventBlockchainSync.sh: Permission denied on LCD screen

davidepatti, 2021-12-22:
I updated to 1.71 using patch menu, and I receive the above message on the LCD.
I think it just repeats until the chain get synced.
This is probably related to a missing exec permission on some setup.scripts, as also reported in:

https://www.reddit.com/r/raspiblitz/comments/qbdvb8/i_tried_to_update_bitcoin_core_on_my_raspiblitz/

openoms, 2021-12-23:
You will.need to reflash thw SDcard to update. See the FAQ.

davidepatti, 2021-12-23:
yes, I know that "release" updates requires reflashing. But in this case I was referring to the "UPDATE->PATCH" option of raspiblitz, which is described just as a way to sync with current github version to fix bugs.
So, I think those missing execution permissions are actually a bug in 1.7.0 patch menu.

rootzoll, 2021-12-23:
"UPDATE > PATCH" is not an update mechanism between versions. It can be used if your are for sure on the lastest version & just a quick security patch was released on the GitHub for it. For all other updates beside quick security/bug fixes you need to flash a new sd card. Also in your situation you should now go the "UPDATE > RELEASE" and flash a v1.7.1 sd card.

davidepatti, 2022-01-06:
anyway, I solve by manually adding the missing permission with a chmod a+x *

closed 2022-01-06 by davidepatti