raspiblitz/raspiblitz #2837: 7 Zoll DSI LCD Display 800x480 fullscreen Support

3773, 2021-12-22:
Hi there,

[7 inch Touchscreen](https://www.amazon.de/gp/product/B096KL11X8/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1)

I bought this 7 inch touchscreen from Amazon and connected it via the DSI interface. Everything works so far, including the touch. Unfortunately, I don't have a full picture of the RaspiBlitz. Can I adjust the screen in full screen mode?

Image
[image]

openoms, 2021-12-23:
Try adjusting the font size. There is an old guthub issue about it, but a general Raspberry Pi method could work too.

blindnew23, 2022-11-29:
hey I couldn't make it work, how did you make the connections?

3773, 2022-12-06:
...from Rasperry 4 per DSI cable to Touchscreen...under Settings "Touchscreen" enable, reboot an works...but very small - not fullscreen - see foto