raspiblitz/raspiblitz #2839: CL seed not appears from menu after recovering

Tracachang, 2021-12-22:
When creating a new CL wallet, seed can be seen from CL menu -> Seed.

I have installed a fresh new raspiblitz and I have recovered CL with my old seed, all works as intended but in CL menu -> Seed it does not shows the seed.
[image]

I can see the seed with `sudo cat /home/bitcoin/.lightning/bitcoin/seedwords.info` since wallet is not encrypted but not from the CL menu

Tracachang, 2021-12-22:
When typing `sudo cat /home/bitcoin/.lightning/bitcoin/seedwords.info`. It shows:

`# This file was placed by cl.hsmtool.sh .
#Contains the seed words from which the hsm_secret in the same directory was generated
seedwords='word word word word word word word word word word word word word word word word word word word word word word word word'
seedwords6x4=''
#Will be removed safely when the hsm_secret is encrypted.
`

While in a new wallet created (appearing seed in CL-> Seed, the seedwords.info file looks like this (appearing seed twice, in seedwords= and in seedwords6x4:

`# This file was placed by cl.hsmtool.sh .
#Contains the seed words from which the hsm_secret in the same directory was generated
seedwords='word word word word word word word word word word word word word word word word word word word word word word word word'
seedwords6x4=''word word word word word word word word word word word word word word word word word word word word word word word word'
#Will be removed safely when the hsm_secret is encrypted.
`

rootzoll, 2021-12-23:
@openoms can you take a look

openoms, 2021-12-23:
thanks for reporting. the problem is the `cl.backup.sh` does not pass the `seedwords6x4` format: https://github.com/rootzoll/raspiblitz/blob/6cd17ad27a901cc852fc734bd23500c02df02d57/home.admin/config.scripts/cl.backup.sh#L440
due to fix

rootzoll, 2022-02-17:
OK closing this issue for v1.7.2 release.

closed 2022-02-17 by rootzoll