raspiblitz/raspiblitz #2842: build continuous integration

m00ninite, 2021-12-23:
Would be nice to have nightly builds, or at the very least, a build that doesn't rely on somebody running it by hand. Longer term goal is to actually deploy the image to a real raspi (or emulator) and test some basic functions. (outside the scope of this initial ticket)

I'm fairly competent with gitlab-ci, so I'm thinking I'll go with that. Need to figure out how well the free tier integrates with github.
GitHub actions is another option, but you can't self-host the runners with GitHub actions, which means somebody would need to pay for it. If there's any input regarding what CI system to use, I'd love to hear it.

I'm actively working on this, so please assign to me :)

GoetzGoerisch, 2021-12-23:
@m00ninite I would assist. GH-Actions are available for Open Source projects, at the moment free of charge.
@rootzoll @openoms please add your suggestions?

openoms, 2021-12-23:
Great initiative! I don't have experience with setting up CI yet, but looking forward to learn.

The first step would be for a nightly build is to set up a 64bit RPi (aarch64) emulator with our base image and run the build_sdcard.sh. The base image source is linked at the top of the script.
At some point this could be built as the part of this repo, but we but there yet.

Then we could test if it boots at all and if it runs various things without errors. This would include writing tests. First only to check for if installs scripts could finish for example.

Running bitcoin and LN related tests would include setting up a regtest instance I presume.

m00ninite, 2021-12-23:
> @m00ninite I would assist. GH-Actions are available for Open Source projects, at the moment free of charge.
Is it totally free? I thought it was free for <x> minutes, or <y> mb a month. Given the size of the project, we'd sail right over the limits rather quickly

> The first step would be for a nightly build is to set up a 64bit RPi (aarch64) emulator with our base image and run the build_sdcard.sh. The base image source is linked at the top of the script.
At some point this could be built as the part of this repo, but we but there yet.

I have some thoughts on this. I'm trying to avoid using any actual hardware or a full-blown emulator, as it simplifies things significantly. I've made some pretty good progress referencing [this gist](https://gist.github.com/jkullick/9b02c2061fbdf4a6c4e8a78f1312a689). Everything appears to build fine, but when I flash the resulting image to an SD card and boot my pi, it brings me to the normal raspbian desktop. Are there any steps missing between running ./build_sdcard.sh, blitz.preparerelease.sh, and uploading the image to github?

GoetzGoerisch, 2021-12-23:
GitHub is here supporting Open Source very well, all public repos are free.

rootzoll, 2021-12-23:
I would also favor GitHub actions ... from what I read only private repos will get billed when reaching limits:
```
GitHub Actions usage is free for both public repositories and self-hosted runners. For private repositories, each GitHub account receives a certain amount of free minutes and storage, depending on the product used with the account. Any usage beyond the included amounts is controlled by spending limits.
```
https://docs.github.com/en/billing/managing-billing-for-github-actions/about-billing-for-github-actions

@m00ninite try the image already after `build_sdcard.sh` - if it still boots to raspbian desktop there is something to fix before that. You build the sd card with `headless` parameter right?

m00ninite, 2021-12-23:
> GitHub is here supporting Open Source very well, all public repos are free.

Looks like I misread, good news!

> @m00ninite try the image already after build_sdcard.sh - if it still boots to raspbian desktop there is something to fix before that. You build the sd card with headless parameter right?

Indeed, I build the SD with `headless` param and then plug it right into my pi, but the raspian desktop still shows over HDMI, and the ssh server isn't running. Any ideas where the build might be going wrong?

hashes4merkle, 2022-07-03:
@m00ninite - hows progress on this workstream?

hashes4merkle, 2022-07-07:
@rootzoll - Currently researching a few different solutions to a CI/CD pipeline.  I've selected [pi-gen](https://github.com/RPi-Distro/pi-gen) to automate builds locally - Initial poc has been successful.  There are some gaps that need to be addressed but nothing too alarming at the moment.

For my next steps, I will test the image on a pi then review a GitHub action from the marketplace that uses pi-gen. [Pi-gen-action](https://github.com/usimd/pi-gen-action) appears stable.  From here, we should have a functioning pipeline to roll nightly builds.

hashes4merkle, 2022-07-28:
Update: POC with [pi-gen](https://github.com/RPi-Distro/pi-gen) is complete.  After discussion with @openoms, I also tested packer + [packer-builder-arm](https://github.com/mkaczanowski/packer-builder-arm) as a potential solution to a CI/CD pipeline.  I was successful in modifying a RaspiOS image to include all artifacts from [`build_sdcard.sh`](https://github.com/rootzoll/raspiblitz/blob/v1.7/build_sdcard.sh).  

Next steps:
- Validate all build steps
- [Build regtest - issue #935180367](https://github.com/rootzoll/raspiblitz/issues/2391#issue-2391)
- Decision location of runner
- Decision storage of image artifacts

openoms, 2022-11-24:
Generated amd64 images for VM use successfully here: https://github.com/openoms/raspiblitz/actions
If anyone knows how to make a bootable disk out of it please comment: https://github.com/rootzoll/raspiblitz/issues/3053#issuecomment-1326674616

closed 2022-12-15 by rootzoll