raspiblitz/raspiblitz #2849: Message "statusscan blocked (124) - fallback to old results" while Syncing

Somativa, 2021-12-24:
I have a problem. from 90% suddenly this message went off and the syncing stops felt.
[image]
"statusscan blocked (124) - fallback to old results" whats this?

Somativa, 2021-12-24:
I saw the other post. But there is no solution in it. Why does this happen and how can I fix it?

marinojoey, 2021-12-28:
I had this same error message. Not sure if it will work for everyone, but my issue was that my network security was blocking it as a potential threat. I allowed an exception for its IP address and the error disappeared and the blockchain downloaded.

rootzoll, 2022-01-04:
> "statusscan blocked (124) - fallback to old results" whats this?

`statusscan blocked` is due to the way the v1.7.1 still gathers system information on the pi before showing status information. This will change with v1.7.2 which introduces a non-blocking background scan for system information.

Why you experience this on "syncing" I dont know. Last resort would be to `CTRL+C` to exit to terminal - use command `restart`. If that still blocks `CTRL+C` to exit to terminal - use command `off` and then startup with a fresh flashed sd card & choose setup fresh (with use existing blockchain data).

davterra, 2022-01-18:
> `statusscan blocked` is due to the way the v1.7.1 still gathers system information on the pi before showing status information. This will change with v1.7.2 which introduces a non-blocking [...]

I am having the same issue after months of rock-solid/no-problems operation. For me, the trouble began after my network switch failed while I was on extended holiday, resulting in my raspiblitz running for weeks but receiving no data.
I've tried re-flashing the SD card, and issue still exists. 
I am ready to try your last suggestion: re-flashing + choosing fresh setup. 
Question: If I go to the main menu and select Repair > Reset HDD is that the same as choosing a fresh setup on installation?
Also, what exactly gets deleted during this....I assume my wallets & onion addresses, etc. remain intact and unchanged?

rootzoll, 2022-01-19:
Beware that "fresh setup" deletes wallets - if you have funds on that node other options should be considered.

If you want to do a "fresh setup" you dont need to go to "Repair" first. You just flash a new sd card, boot your Raspi with that. It will ask you to ssh in again and when asked to recover you just say "other options" and then from the menu "fresh setup". It will then ask you if you want to use the blockchain and delete all other data (including old wallets, etc).

davterra, 2022-01-19:
> Beware that "fresh setup" deletes wallets - if you have funds on that node other options should be considered.
>
> If you want to do a "fresh setup" you dont need to go to "Repair" first. You just [...]

Thanks for that detail. I was considering doing the "repair" option before re-flashing, just to see if that alone would fix the issue.

closed 2022-07-04 by openoms