raspiblitz/raspiblitz #2850: zerotier broken on my raspiblitz

KainerW, 2021-12-25:
Hi, maybe someone can help me here.

I can not get zerotier to work anymore, it stopped from one day to the other. 

Trying to activate it on the raspiblitz-menue I just get a screen with the error mesage "Zerotier join failed".

When I try this: sudo zerotier-cli join [my.zerotier-id]
I do get this:
/usr/sbin/zerotier-cli: 1: /usr/sbin/zerotier-cli: Syntax error: “(” unexpected

Any idea how to fix it or what might cause this error?

openoms, 2021-12-25:
haven't experienced this, but use:
`sudo zerotier-cli`
`--help` is the a useful option
and I use:
`sudo zerotier-cli status`
and
`sudo zerotier-cli listnetworks`
the most often.

KainerW, 2021-12-26:
THX Openoms. I tried but for each of this lines I get back: '/usr/sbin/zerotier-cli: 1: /usr/sbin/zerotier-cli: Syntax error: "(" unexpected'

openoms, 2021-12-26:
try to restart the node and maybe use another terminal to log in.

if that doesn't help:
try reinstalling zerotier from the menu or remove with:
`sudo apt uninstall zerotier`
 and follow the official download instructions from zerotier.com for Debian.

KainerW, 2021-12-26:
I tried this before. 'Sudo apt uninstall zerotier' does not work. It results in 'E: Invalid operation uninstall'. 

Using "sudo apt remove zerotier" results in:
Reading package lists... Error!
E: Encountered a section with no Package: header
E: Problem with MergeList /var/lib/dpkg/status
E: The package lists or status file could not be parsed or opened.

Could I remove the status file and re-generate it somehow without destroying my installations on this blitz/pi?

openoms, 2021-12-26:
It is a weird problem.
can try
`sudo apt remove zerotier* --purge`

otherwise a reflash will solve this. Best to use a new SDcard so rule out the file corruption because of the card failing.

KainerW, 2021-12-26:
sudo apt remove zerotier* --purge gave:

Reading package lists... Error!
E: Encountered a section with no Package: header
E: Problem with MergeList /var/lib/dpkg/status
E: The package lists or status file could not be parsed or opened.

Will try a reflash and report. Thank you.

KainerW, 2021-12-27:
I reflashed (and updated to 1.7.1) a new sd card - zerotier finally worked after restarting the node 3-4 times finally. 

... But now all but one other apps are broken. Only app still working is specter. Sphinx broke, thunderhub broke, mempool broke. Very frustrating.

openoms, 2021-12-27:
Without details nothing can be done.

Look at the output of:
`debug`
yourself and can post a link of the uploaded logs.
Find the command at the end of the debug output.

KainerW, 2021-12-27:
Thank you! -> https://termbin.com/0aa0

Update so far: I have restarted the node about 10-20 times, I have tried/made all updates of all apps via raspi-menu, changed nothing.
Had then to manually delete mempool app and install new to make this app work. THIS fixed mempool-app eventually. 
Same did work for thunderhub - but: I had to try to install it 5 times. 4 times I got error message, then it finally worked.
Sphinx chat app still broken and not working, deinstalled many times, reinstalled many times, updated many times, restarted node many times. still not working.  Maybe you can see what is wrong in the logfile?

openoms, 2021-12-27:
your bitcoind and LND working normally. Can't see the reason for other apps breaking (the logs are not log enough.)

Definitely shouldn't need to restart the node or reinstall apps and this doesn't happen with others, neither I could reproduce.

Did you use a new SDcard when reflashed?
Something has corrupted your system to start with.

You can see get the logs of the individual services in the SYSTEM menu.
eg. for sphinxrelay.

KainerW, 2021-12-27:
Used a new SDcard yes. 
I might get a new pi/hdd as this one seems to be broken/unreliable.

openoms, 2022-01-02:
closing then original issue is fixed

closed 2022-01-02 by openoms