raspiblitz/raspiblitz #2851: Bullseye Problems

rootzoll, 2021-12-26:
Go report that the new Bullseye base image might cause some trouble ...

- SSH service does not start reliebale (but this can also be related to changes in the ssh shell scripts)
- hardware support: problems getting the touchscreen to run

TODO: retest touchscreen

openoms, 2021-12-26:
see my experience on with ssh on Armbian Buster: https://github.com/rootzoll/raspiblitz/issues/2820#issuecomment-1001242865

rootzoll, 2022-01-04:
@fusion44 does this https://github.com/rootzoll/raspiblitz/issues/2820#issuecomment-1001242865 reflects the error you observed?

rootzoll, 2022-01-26:
OK Touchscreen restested - works as it is fo far on my tests.
So its just down to the SSH issue that is addressed in #2820
Closing this issue.

closed 2022-01-26 by rootzoll