raspiblitz/raspiblitz #2854: Raspiblitz not syncing - Cannot reach the SSH Main Menu to initiate UPDATE - Can I just use Etcher to create an up-to-date version? Will I lose my lightning or btc wallet?

OurSiDEWiND, 2021-12-28:
Hello all!

The title pretty much sums up my inquiry. My Raspiblitz is not syncing, I am connected to the internet (successfully pinged yahoo.com from the PI) yet connections are 0 and even if I manually connect to an online node connections stay at 0.

I think updating my Pi would fix this, but because it is on the sync page I cannot press the update button on the SSH Main Page. What are the downsides of just burning the fresh version of raspiblitz onto my current SD card and thus earsing my current one? Will I lose my lightning wallet? my btc wallet? I assume I will lose my passwords.

Your help is greatly appreciated.
Merry Christmas and a Happy New Year to you all

openoms, 2022-01-02:
You can always flash the latest image on the SDcard - this is the same as the update process.
The wallets and unique data are stored on the disk and will be recovered if intact.

cstenglein, 2023-06-07:
Seems resolved? => Closing for now.