raspiblitz/raspiblitz #2855: A few adjustments required to enable LND Hybrid Mode on raspiblitz

TrezorHannes, 2021-12-29:
Went through [the guide](https://github.com/blckbx/lnd-hybrid-mode#configuring-hybrid-mode) to enable Hybrid Mode with my Raspiblitz Node, and encountered a few issues, outlined below:

### lnd.conf
Writing to lnd.conf directly to adjust
1. external port (optional)
2. tor.streamisolation=false
3. tor.skip-proxy-for-clearnet-targets=true

requires at least further bash script adjustments to achieve 1 and 2.
### 1
See [issue reported here](https://github.com/rootzoll/raspiblitz/issues/100#issuecomment-1002658348), the port change script `/home/admin/config.scripts/lnd.setport.sh` checks whether Tor is running, and if true, aborts. Hybrid mode makes this check-function obsolete, or rather needs more delicate checks
FWIW, I had to change the port since my other node already occupies 9735 for Hybrid Clearnet.

### 2
Hybrid Mode and tor.streamisolation=true causes LND daemon to abort and not complete start-up successfully. This is reported by LND systemctl as streamisolation and tor.skip-proxy-for-clearnet-targets=true being incompatible. However, as [suggested here](https://github.com/rootzoll/raspiblitz/issues/237#issuecomment-530671474), `/etc/systemd/system/lnd.service` calling `/home/admin/config.scripts/lnd.check.sh` with a hard-coded streamisolation=true:
line 196 > `setting ${lndConfFile} ${insertLine} "tor.streamisolation" "true"`

### Workaround
I have manually adjusted both and successfully booted LND with Hybrid mode, by
- uncommenting the Tor check in /home/admin/config.scripts/lnd.setport.sh
- set streamisolation=false in /home/admin/config.scripts/lnd.check.sh

openoms, 2022-01-02:
Related discussion in https://github.com/rootzoll/raspiblitz/issues/2787

nerrixDE, 2022-01-05:
Hi, are the steps above working for v1.7.1 with LND 0.13.3-beta (using Wireguard not OpenVPN, but that should not change much)

TrezorHannes, 2022-01-13:
> Hi, are the steps above working for v1.7.1 with LND 0.13.3-beta (using Wireguard not OpenVPN, but that should not change much)

Yes I would certainly say it's working in your environment, too

rootzoll, 2022-03-09:
@openoms wanna target this for soonish 1.8 or push it to 1.8.1? ... related to #2787?