raspiblitz/raspiblitz #2856: Remote access over Tor

jugi01, 2021-12-29:
I wanted to set up remote access according to the instructions in the FAQ: "How to SSH over Tor?".
But I got the following error message.
And when I enter the Onion address in Tor, I got the "unable to connect" message (below).
It can't be the network, I can load all pages and I also have access to another fullnode via an onion address through Tor.
Thank you for any help.

admin@raspberrypi:~ $ bash /home/admin/config.scripts/internet.hiddenservice.sh ssh 22 22
# Hidden Service for ssh
HiddenServiceDir /mnt/hdd/tor/ssh
HiddenServiceVersion 3
HiddenServicePort 22 127.0.0.1:22
Reloading Tor to activate the Hidden Service...
The Tor Hidden Service address for ssh is:
avvk4r6i55yvvfs6ag74uywjrkh555wp4r43jdr3fz7imgnbu5uxakyd.onion
use with the port: 22
admin@raspberrypi:~ $ torsocks ssh admin@HiddenServiceAddress.onion
1640632590 ERROR torsocks[29509]: General SOCKS server failure (in socks5_recv_connect_reply() at socks5.c:527)
ssh: connect to host hiddenserviceaddress.onion port 22: Connection refused
admin@raspberrypi:~ $ sudo cat /mnt/hdd/tor/ssh/hostname
avvk4r6i55yvvfs6ag74uywjrkh555wp4r43jdr3fz7imgnbu5uxakyd.onion
admin@raspberrypi:~ $
 
 
Unable to connect
Firefox can’t establish a connection to the server at avvk4r6i55yvvfs6ag74uywjrkh555wp4r43jdr3fz7imgnbu5uxakyd.onion.
    The site could be temporarily unavailable or too busy. Try again in a few moments.
    If you are unable to load any pages, check your computer’s network connection.
    If your computer or network is protected by a firewall or proxy, make sure that Tor Browser is permitted to access the Web.

openoms, 2022-01-02:
Never share you unique hidden service for your ssh connection - so change it in this case by:
``` 
# deleting the keys;
sudo rm -rf /mnt/hdd/tor/ssh/
# reload tor
sudo systemctl reload tor@default
```

It is likely a local Tor issue and might resolved with a restart.

cstenglein, 2023-06-07:
@jugi01 is this still relevant? Closing for now, if it's still relevant, please re-open

closed 2023-06-07 by cstenglein