raspiblitz/raspiblitz #2858: error initializing LND - (LND service is not running.)

GabrielRCL, 2021-12-30:
What's up guys, how's it going? will someone be able to help me?
my raspiblitz turned off overnight by itself and now doesn't want to turn on LND, what could I do?
[image]

already restarted 2x and still the error continues. raspiblitz makes an automatic backup right? how can i use this automatic backup?
I already did the repair and I have the backup, what can I do to fix this problem without losing my channel funds?

DEBUG:
https://termbin.com/e6cb

GabrielRCL, 2021-12-31:
i have SEED, LND_Rescue (Backup-Repair-raspblitz), and i have Backup_All_Channels (created in RTL).
should i burn a new image to the microSD?
do i need to save something first?
I have never backed up before and am scared of losing my funds. Could someone explain to me how I can recover my channels?

openoms, 2022-01-02:
The channel.db is corrupted.
Please close if you have been able to recover the funds from seed + SCB.

closed 2022-01-02 by openoms