raspiblitz/raspiblitz #2859: Problems with external SWAP device

rootzoll, 2022-01-02:
Recently there are more reports of SWAP device errors. This seems to be because the SWAP file is set by RaspiBlitz to the external SSD ... which seems to cause trouble - bringing the device to a hold in some cases. 

If you encounter this problem (you see SWAP error on device and SSh login is not working anymore) -> try to remove & reconnect power and then let reboot. If you run more than once into this error, you can go after reboot to terminal and run the following command to turn external SWAP off: `sudo /home/admin/config.scripts/blitz.datadrive.sh swap off`

The v1.7.2 should at least make it easier to battle this problem with `sudo /home/admin/config.scripts/blitz.datadrive.sh status` giving more info on the swap file and make External Swap file switchable on SETTINGS.

openoms, 2022-01-07:
Just got this happening repeatedly too.
Not sure what is the cause. HW was stablebso far. The SSD is fairly new and has no smartctl issues.

[image]

rootzoll, 2022-01-07:
Yeah somehow I saw it also more recently on some test devices, not sure what changed .. @openoms do you think its safer to release next sd card image with SWAP on sd card as default (and just create on SSD if switched on explicitly)? I can run some tests if this slows down initial blockchain sync, etc.

rootzoll, 2022-01-07:
Also found this article on the topic:
https://www.makeuseof.com/fix-read-error-on-swap-device-ubuntu/

`The Read-error on Swap-device error occurs when your Ubuntu system is low on swap memory.`

So it could be that some applications increased the RAM use and now the swap file is "too small".
At the moment we have 2GB swap file - but also the Pi4 got more RAM in general .. so what is the best SWAP size?

Going with this article:
https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/storage_administration_guide/ch-swapspace20212
[image]

I think 4GB should be minimum .. here for v1.7.2

openoms, 2022-01-07:
Agree that increasing the swap size should be a good move especially with LND requiring a lot of it sometimes.
As an example when a watchtower is on, hut offline, the state changes are stacked into memory until it can be uploaded when the tower comes online again [lnd issue to be referred here]

As a rule of thumb on my linux laptop I have set twice the RAM for swap so it only slows down, but not freezes.

4 GB on the RPi4 sounds good and a significant enough change to solve the issue. Will change back to the increased size on the disk and see if the problem appears again.

openoms, 2022-01-08:
Setting the swap higher was not straightforward:

with 
```
#CONF_MAXSWAP=2048
```
in `/etc/dphys-swapfile`

it was still limited to 2048:
```
sudo /home/admin/config.scripts/blitz.datadrive.sh swap on

# RASPIBLITZ DATA DRIVES - SWAP FILE
# Switch off/uninstall old SWAP
# Rewrite external SWAP config for EXT4 setup
# Creating SWAP file ..
4096+0 records in
4096+0 records out
4294967296 bytes (4.3 GB, 4.0 GiB) copied, 216.839 s, 19.8 MB/s
# Activating new SWAP
Setting up swapspace version 1, size = 4 GiB (4294963200 bytes)
no label, UUID=44e6a819-f997-4149-9780-82628b28c312
computing size, want /mnt/hdd/swapfile=15744MByte, restricting to config limit: 2048MBytes, checking existing: deleting wrong size file (4294967296), generating swapfile ... of 2048MBytes
# OK - Swap is now ON external
```

Increasing to:
``` 
CONF_MAXSWAP=8000
```
made it to create an 8GB swapfile:
```
# Activating new SWAP
Setting up swapspace version 1, size = 4 GiB (4294963200 bytes)
no label, UUID=13713e3e-67c9-455f-b0fc-17585ea3e1b9
computing size, want /mnt/hdd/swapfile=15744MByte, restricting to config limit: 8000MBytes, checking existing: deleting wrong size file (4294967296), generating swapfile ... of 8000MBytes
```

Setting
```
MAXSWAP=4096
```
did it:
```
# Activating new SWAP
Setting up swapspace version 1, size = 4 GiB (4294963200 bytes)
no label, UUID=1aaf689f-0b5f-471a-b7ad-7c07c85ccbfe
computing size, want /mnt/hdd/swapfile=15744MByte, restricting to config limit: 4096MBytes, checking existing: keeping it
# OK - Swap is now ON external
```

openoms, 2022-01-08:
The node continued to have issues. Froze twice with the last line syslog at :
```
 /home/admin/_bootstrap.sh: line 792: [: -eq: unary operator expected
raspberrypi systemd[1]: systemd-fsckd.service: Succeeded.
raspberrypi _bootstrap.sh[974]: # RASPIBLITZ DATA DRIVES - CLEANING
```
and
```
raspberrypi _bootstrap.sh[966]: /home/admin/_bootstrap.sh: line 792: [: -eq: unary operator expected
raspberrypi _bootstrap.sh[966]: # RASPIBLITZ DATA DRIVES - CLEANING
raspberrypi _bootstrap.sh[966]: # Making sure 'secure-delete' is installed ...
raspberrypi systemd[1]: systemd-fsckd.service: Succeeded.
```

This line bootstrap seems to run without issues when tried separately: 
```
$ sudo config.scripts/blitz.datadrive.sh clean temp
# RASPIBLITZ DATA DRIVES - CLEANING
# Making sure 'secure-delete' is installed ...

# IMPORTANT: No 100% guarantee that sensitive data is completely deleted!
# --> Dont resell or gift data drive. Destroy physically if needed.

# Deleting the temp folder/drive (keeping SWAP file) ..
# OK cleaning done.
```
```
$ sudo swapon --show
NAME              TYPE SIZE USED PRIO
/mnt/hdd/swapfile file   4G 3.8M   -2
```

will continue to run with swap on the SDcard:
```
sudo  config.scripts/blitz.datadrive.sh swap off
```
```
$ sudo swapon --show
NAME      TYPE SIZE USED PRIO
/var/swap file   4G   6M   -2
```

openoms, 2022-01-08:
So after all the node continued to have read errors even after moving the swap to the SDcard.
[image]
[image]

Ended up cloning to a new SSD successfully.

The increase of the default swap size is still a good move I think so would keep that and continue to run with the 4GB swapfile on the new disk

Takeaways:
* The lack of smartctl errors unfortunately don't rule out that there is an SSD failure.
* The `read-error on swap-device` is what it looks like to be - the disk failing, so best is to proceed to backup (create a MIGRATION file after restart) and move to a new disk asap (can attempt to clone the disk with `dd` on linux or sync fresh in a few days on the new disk).

rootzoll, 2022-01-16:
OK swap file with rc1d is now 4GB by default

```
/mnt/hdd/swapfile file   4G 369.5M   -2
```

openoms, 2022-02-16:
it might be best to set up ZRAM as it takes the load off the disk too: https://github.com/rootzoll/raspiblitz/issues/2905

rootzoll, 2022-02-17:
OK tests with the RC2 look good so far with the new setting. Further optimization on SWAP file to be done in new issue https://github.com/rootzoll/raspiblitz/issues/2905 for v1.8 release. Closing this issue for v1.7.2 release.

closed 2022-02-17 by rootzoll