raspiblitz/raspiblitz #2862: unable to connect Trezor to Specter wallet on Raspiblitz.

dohabee, 2022-01-02:
I have created a multisig Bitcoin wallet using Specter installed on Raspiblitz. I am unable to connect a Trezor HW wallet to Specter either via the Pi's usb port or my laptop's usb port. 

debug log -  https://termbin.com/l830

[image]

openoms, 2022-01-03:
Temporarily you could install and run Specter on your laptop to make this work.

For this issue I can confirm that the Trezor One is not even seen when connected. Looks like it could be a kernel or hardware related issue and not Specter.

My Debian laptop shows the Trezor connected with the same cable:
```
$ lsusb
... Generic SatoshiLabs TREZOR
```

openoms, 2022-05-29:
Just retested this, but couldn't find the solution:
https://t.me/raspiblitz/103855

```
lsusb
```
with admin and also bitcoin user shows the Trezor:
```
Bus 001 Device 012: ID 1209:53c1 Generic SatoshiLabs TREZOR
```
but not recognised in Specter.

ALternatives:
https://docs.specter.solutions/desktop/hwibridge/
or use Sparrow