raspiblitz/raspiblitz #2865: Raspiblitz doesn't enter SSH Main Menu after completing sync

treebeard4, 2022-01-03:
Hi all,

I'm a newbie who bought a Raspiblitz with a pre-synced blockchain a few weeks ago. The initial setup worked fine and the node quickly synced the newest blocks to reach 100.00%. From there it doesn't enter the SSH Main Menu, but remains on the "Node is Syncing" screen. 
Issue #2643 seems to describe a similar problem - my Raspiblitz doesn't switch to the Main Menu at all though. The suggested solution for issue #2643 (running patch and then restart) didn't fix the problem. 
Any ideas what to try next? 
Thanks!

openoms, 2022-01-04:
reflash your SDcard with the latest image as in the readme:
https://github.com/rootzoll/raspiblitz#downloading-the-software

won't need to download the chain again

treebeard4, 2022-01-06:
After reflashing the SDcard, the issue remains unsolved. I've left the Raspi running for 24 hours, it still shows the "Node is syncing" screen with 100.00% progress and between 10 and 13 peers.
Is there anything else I can try next or any other info I can provide?
Thanks!

openoms, 2022-01-06:
press ctrl+c

type:
`menu`
for the menu

check the LNDLOGS in SYSTEM

type:
`debug`
for more logs

treebeard4, 2022-01-16:
Enabling the touchscreen and rebooting the Raspi seems to have solved the issue. Thanks for the help :)

closed 2022-01-16 by treebeard4