raspiblitz/raspiblitz #2869: CL run lightning-cli backup-compact regularly

openoms, 2022-01-06:
The CL database backup can fill up the SDcard, but compacting is efficient:
```
$ config.scripts/cl-plugin.backup.sh backup-compact
/home/bitcoin/.lightning/bitcoin/lightningd.sqlite3
#  Running -cli backup-compact ...
{
   "before": {
      "backupsize": 1798100953,
      "version_count": 198720
   },
   "after": {
      "version_count": 2,
      "backupsize": 130356568
   }
}
```
Went from 1.7 GB to 130MB.
the actual database on the same node:
```
$ sudo du -h /mnt/hdd/app-data/.lightning/bitcoin/lightningd.sqlite3
125M	/mnt/hdd/app-data/.lightning/bitcoin/lightningd.sqlite3
```

Can use
`config.scripts/cl-plugin.backup.sh backup-compact`

or just:

`lightning-cli backup-compact`

openoms, 2022-01-06:
Since CL does not need to stop for this it could be a cronjob installed with the backup plugin.

rootzoll, 2022-02-17:
@openoms do you see this issue well tested enough to close for release?

rootzoll, 2022-02-17:
Perfect. So closing issue for release.

closed 2022-02-17 by rootzoll