raspiblitz/raspiblitz #2870: Selecting certain menu options causes hanging

Epic-Sandwich, 2022-01-06:
I'm using raspiblitz 1.7.2rc1 on a Raspberry Pi 4B, 32GB SD card, 1TB SSD. Sometimes when selecting "LND Wallet Options" from the main menu, raspiblitz will hang and display "get raspiblitz config" until I eventually hit ctrl + C to exit out to the command line. Going back into raspiblitz and trying again will result in the same hanging behavior.

If I then try to reboot from the main menu, it will hang at "stop lightning - please wait .." so again I have to exit out with ctrl + C and reboot from the command line with "sudo shutdown -r now".

This happens multiple times a day. From an initial power up or a reboot, I can initially get into the "LND Wallet Options" as well as successfully reboot from the main menu, but after maybe 5 minutes or as much as a half hour of uptime, it will eventually start doing the above described hanging. Any help diagnosing this problem is appreciated.

openoms, 2022-01-06:
Release candidates are for development purposes. For a stable release use v1.7.1.

There have been changes in the systemd lnd.service which increase the timeout.
A lot has been changed since in the `dev` branch. Let us know if it will persist in the next rc if you want to keep testing

Epic-Sandwich, 2022-01-09:
OK I'll close this.

closed 2022-01-09 by Epic-Sandwich