raspiblitz/raspiblitz #2871: Non-ASCII chars make lndforwardreport script fail

davidepatti, 2022-01-06:
With raspiblitz menu, when selecting LND->lndforwardreport, the list is interrupted with several lines:
`home/admin/config.scripts/lnd.fwdreport.sh: line 99: pubKeyAliasLookup: bad array subscript`

this happens whenever a forwarding to be listed contains a non stricly ascii char in the node name. Some examples:
https://amboss.space/node/02ad6fb8d693dc1e4569bcedefadf5f72a931ae027dc0f0c544b34c1c6f3b9a02b
https://amboss.space/node/035b82a3172c53e206e6a47ea95f17702d78619f384281f43feca13747a8c33896

PS. please notice this is a not so rare situation, given the number thunders, symbols etc present in the node community :)

openoms, 2022-01-10:
I don't have peers currently with emojis in the alias so testing is scarce, but stripping the non ascii characters (with an attempt to convert) with `iconv` should work:

```
lncli getnodeinfo 035b82a3172c53e206e6a47ea95f17702d78619f384281f43feca13747a8c33896 | jq '.node.alias' 
"⚡LightThisNow⚡"

lncli getnodeinfo 035b82a3172c53e206e6a47ea95f17702d78619f384281f43feca13747a8c33896 | jq '.node.alias' | iconv -f utf-8 -t ascii -c
"LightThisNow"

lncli getnodeinfo 02ad6fb8d693dc1e4569bcedefadf5f72a931ae027dc0f0c544b34c1c6f3b9a02b | jq '.node.alias'
"rompert.com🔵"

lncli getnodeinfo 02ad6fb8d693dc1e4569bcedefadf5f72a931ae027dc0f0c544b34c1c6f3b9a02b | jq '.node.alias' | iconv -f utf-8 -t ascii -c
"rompert.com"
```

davidepatti, 2022-01-10:
ok it works now, all forwardings are displayed.
There are still a few "line 99: pubKeyAliasLookup: bad array subscript" at the beginning of the list, one for each forwarding involving a channel that you have previously closed, because  ${pubKeyAliasLookup[$channelInPubKey]} or ${pubKeyAliasLookup[$channelOutPubKey]} can't find the channel.
But that's a minor problem.

closed 2022-02-17 by rootzoll