raspiblitz/raspiblitz #2875: Password C Error

Superhans79, 2022-01-09:
Hello Newbie here an novice so please go easy on me.

I have booted up my LND on a Pi after a while due to my main PC which I use for SSH being broken.

When I try to remote onto my LND wallet it keeps saying my password C is incorrect - This is odd as it never was before - even copied and pasted it so I know its correct.

Anyway - Firstly is there a way I can reset this - also I WANT TO look at the logs but I can never get off the screen which is asking for my password C? Can anyone help - I hope its just me being dumb!

openoms, 2022-01-09:
keep pressing CTRL+c to exit to the terminal
the logs will be in SYSTEM - LNDLOGS

Superhans79, 2022-01-12:
I finally got into my PI raspiblitz command line - im trying to get the logs sorry - I will check in this forum to see if there are any instrucions

Superhans79, 2022-01-12:
Here are the klogs :-)

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.0
chainnetwork: bitcoin / main
 07:29:35 up 4 days, 21:46,  2 users,  load average: 0.26, 0.25, 0.26

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2022-01-05 21:18:00 GMT; 6 days ago
  Process: 1455 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 1460 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1492 (bitcoind)
    Tasks: 18 (limit: 4533)
   CGroup: /system.slice/bitcoind.service
           └─1492 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/…

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Tue 2022-01-11 01:09:41 GMT, end at Wed 2022-01-12 07:29:35 GMT. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-01-12T07:19:57Z Socks5() connect to xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333 failed: host unreachable
2022-01-12T07:20:03Z Socks5() connect to ccjrb6va3j6re4lg2lerlt6wyvlb4tod7qbe7rwiouuapb7etvterxyd.onion:8333 failed: host unreachable
2022-01-12T07:20:09Z Socks5() connect to s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333 failed: host unreachable
2022-01-12T07:21:38Z Pre-allocating up to position 0x500000 in rev02880.dat
2022-01-12T07:21:39Z UpdateTip: new best=00000000000000000005f44ca41e4a377c6f168ae7c75459cfe8d84209a0327d height=718269 version=0x20400004 log2_work=93.281511 tx=702099612 date='2022-01-12T07:20:17Z' progress=1.000000 cache=129.2MiB(818187txo)
2022-01-12T07:22:15Z Socks5() connect to xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333 failed: host unreachable
2022-01-12T07:22:21Z Socks5() connect to ccjrb6va3j6re4lg2lerlt6wyvlb4tod7qbe7rwiouuapb7etvterxyd.onion:8333 failed: host unreachable
2022-01-12T07:22:27Z Socks5() connect to s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333 failed: host unreachable
2022-01-12T07:22:55Z Socks5() connect to gliovxxzyy2rkwaoz25khf6oa64c3csqzjn3t6dodsjuf34w6a6ktsyd.onion:8333 failed: host unreachable
2022-01-12T07:24:23Z Socks5() connect to xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333 failed: host unreachable
2022-01-12T07:24:49Z Socks5() connect to s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333 failed: host unreachable
2022-01-12T07:25:15Z Socks5() connect to gliovxxzyy2rkwaoz25khf6oa64c3csqzjn3t6dodsjuf34w6a6ktsyd.onion:8333 failed: host unreachable
2022-01-12T07:26:44Z Socks5() connect to xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333 failed: host unreachable
2022-01-12T07:26:57Z Socks5() connect to ccjrb6va3j6re4lg2lerlt6wyvlb4tod7qbe7rwiouuapb7etvterxyd.onion:8333 failed: host unreachable
2022-01-12T07:27:03Z Socks5() connect to s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333 failed: host unreachable
2022-01-12T07:27:28Z socket send error Connection reset by peer (104)
2022-01-12T07:27:30Z Socks5() connect to gliovxxzyy2rkwaoz25khf6oa64c3csqzjn3t6dodsjuf34w6a6ktsyd.onion:8333 failed: host unreachable
2022-01-12T07:29:07Z Socks5() connect to xlpi353v7ia5b73msynr7tmddgxoco7n2r2bljt5txpv6bpzzphkreyd.onion:8333 failed: host unreachable
2022-01-12T07:29:14Z Socks5() connect to ccjrb6va3j6re4lg2lerlt6wyvlb4tod7qbe7rwiouuapb7etvterxyd.onion:8333 failed: host unreachable
2022-01-12T07:29:18Z Socks5() connect to s7m4mnd6bokujhywsocxibispktruormushdroeaeqeb3imvztfs3vid.onion:8333 failed: host unreachable

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2022-01-05 21:18:00 GMT; 6 days ago
  Process: 1493 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 1495 (lnd)
    Tasks: 11 (limit: 4533)
   CGroup: /system.slice/lnd.service
           └─1495 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v…

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Tue 2022-01-11 01:09:41 GMT, end at Wed 2022-01-12 07:29:35 GMT. --
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2022-01-05 21:18:01.486 [INF] LTND: Version: 0.12.1-beta commit=v0.12.1-beta, build=production, logging=default, debuglevel=debug
[... 182 lines cut ...]
*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.23'
tempCelsius='55.9'
uptime=423979
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=718269
initialSync=0
syncProgress=100.00 
startcountLightning=1
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

openoms, 2022-01-12:
LND is waiting for the password.
Try the command:
`lncli unlock`
in the terminal and look at the logs again.

LND  Version: 0.12.1-beta is very old, multiple raspiblitz releases behind, would advise an update

Superhans79, 2022-01-12:
Hi

I just keep getting

admin@raspberrypi:~ $ lncli unlock
Input wallet password: 
[lncli] rpc error: code = Unknown desc = invalid passphrase for master public key

It is the correct password too - Is there some instructions to upgrade and can I do that with out the walled password? THanks
SOrry to be a pain

openoms, 2022-01-12:
clearly your passwordC is not matching the wallet file or entered incorrectly (eg. caps lock)

you can upgrade by reflashing the latest SDcard image (see the FAQ), but that won't unlock your wallet or change your passwordC

if you have the seed you can regenerate the wallet file from it and copy in place while keeping all the channels (this is slightly technical)

you would need the seed to recover from seed+SCB as well which will force close the channels with the online peers
To recover from seed and SCB you would need to use the v1.7.0 image (the v1.7.1 does not work for that): https://github.com/rootzoll/raspiblitz/tree/v1.7.0#downloading-the-software

Superhans79, 2022-01-18:
Hello
Well I must of somehow amended ,y password - icant see how though
I am on Version 1.7.0 so are there any instructions for the above? I have the seed

Really appreciate your time

openoms, 2022-01-21:
Here is the path of every file in `/mnt/hdd/lnd`:
```
/mnt/hdd/lnd/tls.cert.old
/mnt/hdd/lnd/v3_onion_private_key.backup
/mnt/hdd/lnd/lnd.conf
/mnt/hdd/lnd/letsencrypt
/mnt/hdd/lnd/v3_onion_private_key
/mnt/hdd/lnd/tls.cert
/mnt/hdd/lnd/tls.key.old
/mnt/hdd/lnd/v2_onion_private_key.backup
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.42679.gz
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.42678.gz
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.42680.gz
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log.1
/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
/mnt/hdd/lnd/logs/bitcoin/mainnet
/mnt/hdd/lnd/logs/bitcoin
/mnt/hdd/lnd/logs
/mnt/hdd/lnd/data/graph/mainnet/sphinxreplay.db
/mnt/hdd/lnd/data/graph/mainnet/channel.db
/mnt/hdd/lnd/data/graph/mainnet/sphinxreplay.db.last-compacted
/mnt/hdd/lnd/data/graph/mainnet/wtclient.db
/mnt/hdd/lnd/data/graph/mainnet/channel.db.last-compacted
/mnt/hdd/lnd/data/graph/mainnet/wtclient.db.last-compacted
/mnt/hdd/lnd/data/graph/mainnet
/mnt/hdd/lnd/data/graph
/mnt/hdd/lnd/data/watchtower/bitcoin/mainnet/watchtower.db
/mnt/hdd/lnd/data/watchtower/bitcoin/mainnet/watchtower.db.last-compacted
/mnt/hdd/lnd/data/watchtower/bitcoin/mainnet
/mnt/hdd/lnd/data/watchtower/bitcoin
/mnt/hdd/lnd/data/watchtower/v3_onion_private_key
/mnt/hdd/lnd/data/watchtower
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/wallet.db
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/macaroons.db.last-compacted
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/invoice.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/signer.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/chainnotifier.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/charge-lnd.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/admin.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/readonly.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/router.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/invoices.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/walletkit.macaroon
/mnt/hdd/lnd/data/chain/bitcoin/mainnet/macaroons.db
/mnt/hdd/lnd/data/chain/bitcoin/mainnet
/mnt/hdd/lnd/data/chain/bitcoin
/mnt/hdd/lnd/data/chain
/mnt/hdd/lnd/data
/mnt/hdd/lnd/tls.key
/mnt/hdd/lnd/delete.txt
/mnt/hdd/lnd
```

You can do something like 
(WARNING : I DID NOT TEST THIS - MAKE SURE YOU UNDERSTAND WHAT IS BEING DONE):
```
# make sure you have the seed and the latest channel.backup saved off the blitz just in case

# stop lnd
sudo systemctl stop lnd

# make a copy of the lnd directory
sudo cp /mnt/hdd/lnd /mnt/hdd/lnd.copy

# move the locked wallet.db
sudo mv /mnt/hdd/lnd.copy/data/chain/bitcoin/mainnet/wallet.db /mnt/hdd/lnd.copy/data/chain/bitcoin/mainnet/wallet.db.locked

# now on v1.7.1 reset and recover your wallet from seed and set your new password C
# finish the process, reboot, unlock etc
# once synced stop lnd again
sudo systemctl stop lnd

# copy back the old data (remember the old wallet was renamed)
sudo cp /mnt/hdd/lnd.copy /mnt/hdd/lnd

# fix the owner
sudo chown -R bitcoin:bitcoin /mnt/hdd/lnd

# reboot
restart

# should be able to unlock with you new password C and hopefully the channels will be back
```

openoms, 2022-04-12:
fixed in v1.7.2

closed 2022-04-12 by openoms

techric, 2024-12-09:
Hello ... 

I've seemed to have come across the same issue. 

I've reflashed my microSD to the latest version of the full Raspiblitz software. 

I reset "Password A" but I'm stuck on the "Password C" unlock screen. I've tried to go to the CLI by pressing on Ctrl+C and Ctrl+Z but I cannot get to the CLI to do trouble shooting. 

Please advise. 

Thanks!

rootzoll, 2024-12-12:
try some random password and the whilel it tries to unlock - use CTRL+C and CTRL+Z to exit to terminal for further debug or deactivation/reset of your Lightning wallet.