raspiblitz/raspiblitz #2877: tor@.service / tor@service failing

openoms, 2022-01-10:
There is a failed service:
```
₿ systemctl --type=service | grep tor@
  tor@default.service                                                                               loaded active     running      Anonymizing overlay network for TCP
● tor@service.service                                                                               loaded failed     failed       Anonymizing overlay network for TCP (instance service)
```
fixed by: 
```
sudo systemctl disable tor@service
sudo systemctl reset-failed tor@service
```

I wonder it should not be enabled in https://github.com/rootzoll/raspiblitz/blob/2bb53ceb9626c3d4c43c4cf7f76574db9dc7f6e9/home.admin/config.scripts/tor.install.sh#L221
what do you think @nyxnor ?

nyxnor, 2022-01-10:
Yes, reset-failed is a good idea, ACK

```sh
  # enable tor services
  sudo systemctl unmask tor@default
  sudo systemctl reset-failed tor@*
  sudo systemctl daemon-reload
  sudo systemctl enable --now tor@ tor@service
  sudo systemctl restart tor@default
```

 reset-failed [PATTERN...]
      so `tor*` would work

openoms, 2022-01-10:
that is just  a temporary fix until restart. I think  `tor@` should not be enabled at all as it is not part of the default Tor installation either.

nyxnor, 2022-01-10:
no problem, remove it pls.
just discoverede now it is for instances.

rootzoll, 2022-01-17:
On a test recovery on LIT its exiting because of an torsocks error @nyxnor @openoms has anyone of you seen this before?

```
Jan 17 09:27:38 raspberrypi torsocks[4109977]: could not load config: could not read UI password: mandatory password for UI not configured. specify either a password directly or a file or environment variable that contains the password
```

openoms, 2022-01-17:
@rootzoll this is a litd error just appears with torsocks because the binary is run with tor because of the Lightning Terminal OCnnect would leak the IP otherwise.

Looks like the config does not have the PasswordB placed as the password for the GUI.

rootzoll, 2022-01-17:
OK LIT looks good now.

openoms, 2022-02-06:
still there on recovery with latest ceommit on `dev`:
```
sudo systemctl list-units --failed
  UNIT                LOAD   ACTIVE SUB    DESCRIPTION
● tor@service.service loaded failed failed Anonymizing overlay network for TCP (instance service)

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.
1 loaded units listed.
```

openoms, 2022-02-09:
```
₿ sudo systemctl list-units --failed
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed.
```

closed 2022-02-17 by rootzoll