raspiblitz/raspiblitz #2878: v1.7.0 Node can't backup - inactive channels - can't upgrade

mantiev, 2022-01-10:
I'm having an issue with my node. Channels that I have will go inactive for no reason, even when the peer is still up. They see me as inactive and I see them as inactive. To make things worse, I can't upgrade my node. When I try to do a backup the node gets caught in a boot loop and I have to restart the node. Same behavior when I try the update path from the node. I can't get it past doing a backup. Any assistance would be greatly appreciated. I'm having channels closed because they think my node is inactive when it's not.

openoms, 2022-01-10:
follow the regular update process: flash a new SDcard with the v1.7.1 image and boot
keep the old one for backup (unless you made sure you have a latest copy of the channel.backup)

openoms, 2022-07-04:
closing for inactivirty

closed 2022-07-04 by openoms