raspiblitz/raspiblitz #2880: migratoin from umbrel to raspiblitz fails?

kanutoson, 2022-01-11:
I wanted to migrate from Umbrel to raspiblitz today. Everything went smoothly up to this window.

[image]

I made a report about `debug`

```

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.1
chainnetwork: bitcoin / main
 13:32:22 up 55 min,  2 users,  load average: 0.39, 0.55, 0.60

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin-Daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2022-01-11 12:38:08 GMT; 54min ago
  Process: 2922 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 2924 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
 Main PID: 2987 (bitcoind)
    Tasks: 18 (limit: 4532)
   CGroup: /system.slice/bitcoind.service
           └─2987 /usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log

Jan 11 12:37:54 Orbi7 systemd[1]: Starting Bitcoin-Daemon...
Jan 11 12:38:08 Orbi7 systemd[1]: Started Bitcoin-Daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Tue 2022-01-11 12:36:47 GMT, end at Tue 2022-01-11 13:32:22 GMT. --
Jan 11 12:37:54 Orbi7 systemd[1]: Starting Bitcoin-Daemon...
Jan 11 12:38:08 Orbi7 systemd[1]: Started Bitcoin-Daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-01-11T12:40:07Z New outbound peer connected: version: 70016, blocks=718153, peer=4 (outbound-full-relay)
2022-01-11T12:40:15Z New outbound peer connected: version: 70016, blocks=718153, peer=5 (outbound-full-relay)
2022-01-11T12:40:21Z New outbound peer connected: version: 70016, blocks=718153, peer=6 (outbound-full-relay)
2022-01-11T12:40:29Z New outbound peer connected: version: 70016, blocks=718153, peer=7 (outbound-full-relay)
2022-01-11T12:40:59Z New outbound peer connected: version: 70016, blocks=718153, peer=8 (outbound-full-relay)
2022-01-11T12:41:06Z New outbound peer connected: version: 70016, blocks=718153, peer=9 (block-relay-only)
2022-01-11T12:43:33Z New outbound peer connected: version: 70016, blocks=718153, peer=11 (outbound-full-relay)
2022-01-11T12:43:36Z New outbound peer connected: version: 70016, blocks=718153, peer=12 (block-relay-only)
2022-01-11T12:47:21Z New outbound peer connected: version: 70016, blocks=718153, peer=13 (outbound-full-relay)
2022-01-11T12:59:13Z New outbound peer connected: version: 70016, blocks=718153, peer=20 (outbound-full-relay)
2022-01-11T13:01:22Z New outbound peer connected: version: 70016, blocks=718153, peer=22 (block-relay-only)
2022-01-11T13:05:16Z New outbound peer connected: version: 70016, blocks=718153, peer=24 (block-relay-only)
2022-01-11T13:10:03Z New outbound peer connected: version: 70016, blocks=718153, peer=26 (block-relay-only)
2022-01-11T13:10:23Z Potential stale tip detected, will try using extra outbound peer (last tip update: 1890 seconds ago)
2022-01-11T13:11:16Z New outbound peer connected: version: 70016, blocks=718153, peer=27 (outbound-full-relay)
2022-01-11T13:12:47Z UpdateTip: new best=00000000000000000000cbb1cfcb479df8eff1cf7737db910ad60537127bf89e height=718154 version=0x20e00004 log2_work=93.280068 tx=701883888 date='2022-01-11T13:12:12Z' progress=1.000000 cache=3.0MiB(22438txo)
2022-01-11T13:14:11Z New outbound peer connected: version: 70016, blocks=718154, peer=28 (outbound-full-relay)
2022-01-11T13:18:18Z New outbound peer connected: version: 70016, blocks=718154, peer=29 (block-relay-only)
2022-01-11T13:25:40Z UpdateTip: new best=00000000000000000007796418362606db9a6f78ad5195ed6d60f2a8dd271b9f height=718155 version=0x29c9c004 log2_work=93.280080 tx=701887628 date='2022-01-11T13:25:18Z' progress=1.000000 cache=4.9MiB(35899txo)
2022-01-11T13:31:07Z UpdateTip: new best=0000000000000000000568494553e89bf7222b484143fdc7d20358cd59ad9455 height=718156 version=0x37ffe000 log2_work=93.280093 tx=701890646 date='2022-01-11T13:30:39Z' progress=1.000000 cache=6.3MiB(47566txo)

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2022-01-11 13:31:39 GMT; 43s ago
[... 174 lines cut ...]
startcountBlockchain=1
bitcoinActive=1
blockchainPeers=10
blockchainHeight=718156
initialSync=0
syncProgress=100.00 
startcountLightning=54
lndActive=0
clActive=0
CLwalletLocked=0
blitzTUIActive=0
blitzTUIRestarts=0
vagrant=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
debug | torsocks nc termbin.com 9999

```

I only wanted to use the synchronized blockchain, no lighting channels were open.

can someone help me?

rootzoll, 2022-01-11:
Ah OK: `2022-01-11 13:30:38.525 [ERR] LTND: unable to open local channeldb: channel db cannot revert to prior version`

So you were using before a higher LND version on umbrel then is runnig on the v1.7.1 raspiblitz.

For your shortterm fix: Please try "CTRL+C" to exit to terminal and then run command `menu` UPDATE > LND and the choose the highest version possible - even when called reckless. Then restart raspiblitz with command `restart`.

TODO: For v1.7.2 check Umbrel Migration script again, how its possible to prevent Migration if LND version has already jumped major.

AaronDewes, 2022-01-11:
@rootzoll You should be able to get the LND version like this:

`grep "lightninglabs/lnd" /mnt/hdd/umbrel/docker-compose.yml | sed 's/.*lnd://' | sed 's/@.*//'`

Example output: `v0.14.1-beta`

For Citadel,

`grep "lightninglabs/lnd" /mnt/hdd/citadel/docker-compose.yml | sed 's/.*lnd://' | sed 's/@.*//'`

AaronDewes, 2022-01-11:
For Umbrel < v0.4.1, this doesn't work. In that case, the LND version will be older than he one in RaspiBlitz though.

kanutoson, 2022-01-11:
I have updated LND and now I am asked to enter the password c for the wallet.

The password is not accepted. i am sure i am using the correct password, but of course I cannot rule out a mistake.

Is there a possibility to use a new password without knowing the old one? As I said, there are no coins on the wallet, maybe just create a new one?

rootzoll, 2022-01-11:
@kanutoson try the umbrel default wallet password: `moneyprintergobrrr`

@AaronDewes thanks :) Thats a good way to figure out the lnd version on an umbrel and prevent this situation. Will give it a try.

kanutoson, 2022-01-11:
the default password works!

now everything seems to be working  - thanks!

nyxnor, 2022-01-12:
```sh
grep -oP "lightninglabs/lnd:\Kv[^@ ]*" /mnt/hdd/umbrel/docker-compose.yml /mnt/hdd/citadel/docker-compose.yml
```

rootzoll, 2022-02-17:
tested again and working with umbrel 0.4.14 - closing issue für release.

closed 2022-02-17 by rootzoll