raspiblitz/raspiblitz #2882: Electrum server not starting after changing password B

elo79, 2022-01-13:
Hi there,
bitcoin and lnd node is synced and running.
I decided to cange Password B (via raspiblitz menu), but after the reboot the Elevtrum deamon is not able to start.
I assume because of the password change? Any ideas how to fix that?

`Jan 13 13:05:56 raspberrypi electrs[3088]:     JSON-RPC error: transport error: Didn't receive response data in time, timed out.
Jan 13 13:05:56 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Jan 13 13:05:56 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Jan 13 13:06:56 raspberrypi systemd[1]: electrs.service: Service RestartSec=1min expired, scheduling restart.
Jan 13 13:06:56 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 1.
Jan 13 13:06:56 raspberrypi systemd[1]: Stopped Electrs.
Jan 13 13:06:56 raspberrypi systemd[1]: Started Electrs.
Jan 13 13:06:56 raspberrypi electrs[11428]: Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir:
Jan 13 13:06:56 raspberrypi electrs[11428]: [2022-01-13T13:06:56.636Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics
Jan 13 13:06:56 raspberrypi electrs[11428]: [2022-01-13T13:06:56.687Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 135
Jan 13 13:07:00 raspberrypi electrs[11428]: [2022-01-13T13:07:00.602Z INFO  electrs::chain] loading 718474 headers, tip=000000000000000
Jan 13 13:07:04 raspberrypi electrs[11428]: [2022-01-13T13:07:04.192Z INFO  electrs::chain] chain updated: tip=000000000000000000026912
Jan 13 13:07:34 raspberrypi electrs[11428]: [2022-01-13T13:07:34.480Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/
Jan 13 13:07:34 raspberrypi electrs[11428]: Error: daemon not available
Jan 13 13:07:34 raspberrypi electrs[11428]: Caused by:
Jan 13 13:07:34 raspberrypi electrs[11428]:     JSON-RPC error: transport error: Didn't receive response data in time, timed out.
Jan 13 13:07:34 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, status=1/FAILURE
Jan 13 13:07:34 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Jan 13 13:08:34 raspberrypi systemd[1]: electrs.service: Service RestartSec=1min expired, scheduling restart.
Jan 13 13:08:34 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter is at 2.
Jan 13 13:08:34 raspberrypi systemd[1]: Stopped Electrs.
Jan 13 13:08:34 raspberrypi systemd[1]: Started Electrs.
Jan 13 13:08:34 raspberrypi electrs[20319]: Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir:
Jan 13 13:08:34 raspberrypi electrs[20319]: [2022-01-13T13:08:34.633Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics
Jan 13 13:08:34 raspberrypi electrs[20319]: [2022-01-13T13:08:34.675Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 136
Jan 13 13:08:38 raspberrypi electrs[20319]: [2022-01-13T13:08:38.635Z INFO  electrs::chain] loading 718474 headers, tip=000000000000000
Jan 13 13:08:42 raspberrypi electrs[20319]: [2022-01-13T13:08:42.138Z INFO  electrs::chain] chain updated: tip=000000000000000000026912
Jan 13 13:09:12 raspberrypi electrs[20319]: [2022-01-13T13:09:12.434Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/
Jan 13 13:09:12 raspberrypi electrs[20319]: Error: daemon not available
Jan 13 13:09:12 raspberrypi electrs[20319]: Caused by:
Jan 13 13:09:12 raspberrypi electrs[20319]:     JSON-RPC error: transport error: Didn't receive response data in time, timed out.
`

openoms, 2022-01-13:
Edit:
`sudo nano /home/electrs/.electrs/config.toml`
and put the right passwordB.
then restart electrs.

Fixed in the dev branch fro the new release.

elo79, 2022-01-13:
Wow, that was quick!
That fixed it, thanks.
Continue the great work.

closed 2022-01-14 by openoms