raspiblitz/raspiblitz #2884: BTC Pay Server more robust install script

rootzoll, 2022-01-13:
If on first try the install of BTCPay fails - further install attemts just might do an hard exit:
https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/bonus.btcpayserver.sh#L259

Check how to improve this. Maybe clean up (check user & delete first) on second install?

rootzoll, 2022-01-13:
Also check thunderhub on install robustness in case of net outtages during install.

openoms, 2022-01-13:
Instead of just exiting the `off` option could be called to clean up a previously failed / incomplete installation.

rootzoll, 2022-12-13:
I added fatpack options for all apps that can be installed thru WebUI. This means also BTCPayServer bonusscript has now a seperate `install` & `on` function and `install`that creates user home and installs/compiles the code base is already called during sd card `fatpack` option. So the basic code compile will already be part of the default sd card and well tested on release - while `on` triggered by the user will be much faster now and just edit config and activating the btcpay/nbxplorer systemd services with less fail options - so offering a more overall robust install experience.

This is to be tested in detail during release candidate phase.

closed 2022-12-15 by rootzoll