raspiblitz/raspiblitz #2887: [dev] mempool not installed

adevoss, 2022-01-14:
After starting fresh I saw thunderhub and mempool were not installed.
Due to an internet connection issue on my side during the setup this went wrong.
Manually installed Thunderhub.

And then I executed:
`sudo ./bonus.mempool.sh on`

Result:
A very lengthy error message (~2100 lines). Summary:
```
Error: src/app/app.module.ts:49:225 - error TS2307: Cannot find module '@fortawesome/free-solid-svg-icons' or its corresponding type declarations.
...
FAIL - npm run build did not run correctly, aborting
```

openoms, 2022-01-15:
I have just installed mempool on my X64 Debian build ok.

Try to reproduce and don't use `sudo` with the script. It should be run with `admin`.

Clean with:
`config.scripts/bonus.mempool.sh off`

adevoss, 2022-01-16:
Cleanup gave this:
# Mempool is not installed.

Executed as admin:
...
# npm install for mempool explorer (frontend)
...
mempool version 2.2.2
mempool revision 251a1af4

Two warnings about optimization bailouts
Then the same error.

openoms, 2022-01-16:
what is your:
`node --version`
`npm --version`?

adevoss, 2022-01-16:
node --version
v16.10.0
npm --version
7.24.0

openoms, 2022-01-16:
that is good and matches the requirements of mempool.
There is a new release you could try: https://github.com/mempool/mempool/releases/tag/v2.3.0

openoms, 2022-01-16:
v2.3.0 (just changing the version in the script) installed fine on my baremetal X86 Debian. Could update for v1.7.2.

adevoss, 2022-01-16:
Bare metal or vm shouldn't make a difference in this case I think.

```
config.scripts/bonus.mempool.sh off
pinnedVersion="v2.3.0"
config.scripts/bonus.mempool.sh on
```
But still get this (watch version numbers)
```
fatal: destination path 'mempool' already exists and is not an empty directory.
fatal: ambiguous argument 'v2.3.0': unknown revision or path not in the working tree.
```
...
```
...
> mempool-frontend@2.2.2 build
> npm run generate-config && ng build --prod --localize && npm run sync-assets && npm run build-mempool.js

> mempool-frontend@2.2.2 generate-config
> node generate-config.js

mempool-frontend-config.json file not found, using default config
Copied src/index.mempool.html to src/index.html
mempool version 2.2.2
mempool revision 251a1af4
No configuration updates, skipping generated-config.js file update
...
```
Then same error (or simuilar).

openoms, 2022-01-16:
@adevoss I repeatedly cannot replicate these errors.

running `config.scripts/bonus.mempool.sh off` still left the system in a dirty state.

Use:
`sudo userdel -rf mempool`
or 
`config.scripts/bonus.mempool.sh off`
with the latest commit.

adevoss, 2022-01-16:
Sorry I thought I could choose one of the two.
Now
`sudo userdel -rf mempool`
and
`config.scripts/bonus.mempool.sh off`

I don't know how to get a specific commit so I took it from `dev`.
This made a difference. Now seeing blocks in the browser on port 4080.

openoms, 2022-01-17:
ok, thanks for testing. Closing as it is solved.

openoms, 2022-01-17:
To be clear just `config.scripts/bonus.mempool.sh off` is working since c17b25e

closed 2022-01-17 by openoms