raspiblitz/raspiblitz #2889: Ip2tor subscription issue

jmillionaire, 2022-01-15:
Hi, @openoms, I contacted you on Twitter about trying to re-establish a IP2TOR subscription. I cancelled my subscription, have tried renewing and it times out.

openoms, 2022-01-16:
Are you on the latest release: v1.7.1? 
Are you able to pay other LN invoices?

@rootzoll is there anything to do client side to update an IP2Tor subscription?

frennkie, 2022-01-16:
@jmillionaire sorry for the inconvenience... there was a system crash on ip2tor some days ago. Service were restored but unfortunately the backend LND node was down yesterday - therefor you got those timeouts. Node is back online since 3am and it should work - please try it.

frennkie, 2022-01-16:
Please make sure that you order the bridge correctly - especially if you use the "custom" feature. I have on request from yesterday for: "02...ac@ir..qd.onion:9735:9735" <- this can not work.

jmillionaire, 2022-01-16:
@frennkie i've tried each site (fra, nam) and the process gets stuck at `#### Paying invoice ...` 

anything you can see on your end for invoice: `lnbc280n1ps7...d9hqq5clprc` ?

thanks for your help.

jmillionaire, 2022-01-16:
sorry, just now understood your correction above. tried again trimming the address, it's still stuck on `#### Paying invoice ...`

`# invoice: lnbc300n1ps7...0eqptdvdjv`

jmillionaire, 2022-01-16:
i'm seeing 2 payments completed in RTL now, though, but no active subscription in my Raspiblitz.

jmillionaire, 2022-01-16:
finally worked. thanks!

closed 2022-01-17 by rootzoll