raspiblitz/raspiblitz #2890: 1.7.2rc1d: Does not set hostname

GoetzGoerisch, 2022-01-16:
The lastest RC image added with 4e725f76514b7df75dd2cd8fc229844f2d6445fc does not set the raspiblitz hostname on a new install.

**Steps to reproduce:**
- Download and flash image
- setup new raspiblitz with image

**Expected behavior:**
- hostname is set to raspiblitz node name

**Actual behavior:**
- hostname is still `raspberrypi`

rootzoll, 2022-01-16:
Thats a longer on-going discussion. There is one camp that likes to keep it to `raspberrypi` to have a better op-sec in local networks and another that likes the have the name announced so they can more easily find it in their own network.

At the moment we just have the config value `hostname` in the `raspiblitz.conf` to generally name your raspiblitz. But where to all use that name is not clear. Your lnd or c-lightning node you could change the name by editing the configs. 

So maybe we need to give more options in the config - so that `hostname` really is for the network-hostname and a new `defaultname` to use for naming on other use cases (like on display, install of new lightning impl, etc) .. not sure yet and open for disussion.

@openoms @frennkie @nyxnor ... will add the `teamdiscuss` lable and mark for v1.8 where with the WebUi we will anyway have another place where names will be used.

GoetzGoerisch, 2022-01-16:
`1.7.1` had the behavior if I recall correctly? Therefore, I opend the issue to let you know what I observed during testing of `1.7.2.rc1`.

My opinion would be to have config option during setup, where one sets the new name of the node.

openoms, 2022-01-16:
The name of the node means the alias of the lightning node. 
In my opinion the hostname should not be changed as it could correlate the IP address with the Tor-only LN node at the ISP. See: https://github.com/rootzoll/raspiblitz/issues/819#issuecomment-540084500

To change the actual hostname from a menu can use:
`sudo raspi-config` -> System
[image]
but this won't be preserved with SDcard changes as now.

nyxnor, 2022-01-16:
agree with oms above, that is all.
or modify the hostname directly inside `/etc/hostname`.

GoetzGoerisch, 2022-01-16:
Thanks for the explanation. I'm fine with the current behavior.

openoms, 2022-02-10:
@rootzoll now on v1.7.2rc2b started a new node and the hostname changed to the name given as the node name.
I think as the discussion above this is not the intended behaviour. We should just keep `raspberrypi`.

GoetzGoerisch, 2022-02-10:
I support @openoms comment.

rootzoll, 2022-02-10:
PR merged

closed 2022-02-21 by GoetzGoerisch