raspiblitz/raspiblitz #2891: Add to Install Menue: JoinMarket Web UI

rootzoll, 2022-01-16:
For v1.7.2 the JoinMarket WebUI was merged but just available thru commandline install - see #2747 

For v1.8 update and integrate into SSH and Web one-click install menus.

rootzoll, 2022-05-09:
The RaspiBlitz will start for v1.8.0 with just a limited set of apps to install over WebUI - but JoinMarket is a hot caniidate for a quick follow up 1.8.1 release.

dergigi, 2022-05-19:
> JoinMarket is a hot candidate for a quick follow up 1.8.1 release

Awesome! [v0.0.6](https://github.com/joinmarket-webui/joinmarket-webui/milestone/5) will be quite okay already, and I hope that we'll be having something proper (and almost feature-complete) with [v0.0.7](https://github.com/joinmarket-webui/joinmarket-webui/milestone/6). Getting there!

openoms, 2022-09-28:
Related PR: #3321 

A few more tasks:
- [ ] add Jam to menu and recovery
- [ ] expose the order-book on /obwatch with nginx
- [ ] test

rootzoll, 2022-11-09:
TODO: Make it WebUI ready. @openoms image contect & description needed.

openoms, 2022-11-09:
Referring to https://jamapp.org/

cstenglein, 2022-11-10:
Added issue here as well https://github.com/cstenglein/raspiblitz-web/issues/519

rootzoll, 2022-12-13:
Because of time constrains - pushing this to v1.10

rootzoll, 2022-12-18:
OK Backend & WebUI prepared: https://github.com/cstenglein/raspiblitz-web/pull/545#event-8064122396

ToDo on RaspiBlitzScripts:
- [ ] install/on fatpack
- [ ] status

rootzoll, 2022-12-19:
OK I prepared the installscripts of JAM (and JoinMarket as dependency) for the need to be installed by WebUI - see & test on branch: `2891-webuiapps` - BUT there is one thing thats still a bit tricky:

@openoms I added in that branch the `install` action on the JAM & JoinMarket install scripts. But still the most of the joinmarket install is happening in the `on` section - I think on calling a script from JoinInBox here: https://github.com/rootzoll/raspiblitz/blob/6bd2e38f610aadcb22e02fd03ec37b266eed50cd/home.admin/config.scripts/bonus.joinmarket.sh#L228 .. is there a way to already call just the install/compile without the inital config? So when the WebUI calls the `on` it will not take multiple minutes anymore?

openoms, 2022-12-19:
The install option can call only the install without config with:
```
sudo -u joinmarket /home/joinmarket/install.joinmarket.sh -i install
```

Then the `on` option can do:
```
# configure joinmarket (includes a check if it is installed)
sudo -u joinmarket /home/joinmarket/start.joininbox.sh

# set the raspiblitz.conf
/home/admin/config.scripts/blitz.conf.sh set joinmarket "on"
```

rootzoll, 2022-12-19:
@openoms the `install` section of the joinmarket install script had still a lot of config section that require the HDD/SSD being already available in it - I think I moved most of them over in to the `on` section but I think I still saw some references of the JoinInBox referring to `/home/joinmarket` which later on in the `on` section gets linked to `/mnt/hdd/app-data/.joinmarket` .. are you sure there is no config going on when calling `install.joinmarket.sh -i install`. 

Please during ReleaseCadidate make sure to test all of JAM on the RaspiBlitz fatpack image - and not just on the VMs, etc.

rootzoll, 2023-05-11:
OK Jam will be part of the next v1.9.0 release. Closingh issue.

closed 2023-05-11 by rootzoll