raspiblitz/raspiblitz #2893: Cannot access SD image download

marinojoey, 2022-01-17:
I wanted to upgrade my HDD to an SSD. So I created a migration file and wiped my SD card. I am now attempting to re-write but am running into problems. 

First, this link does not work for me: https://raspiblitz.fulmo.org/images/raspiblitz-v1.7.1-2021-10-28.img.gz so I cannot download the new image.

Second, When I go through the steps for writing my own SD card I run into issues. I tried writing it on the newest arm64 img found here (zip file): https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/ but when I got to the `wget` command (wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh && sudo bash build_sdcard.sh) in the docs It always terminated saying that 'no version of python was installed'. 

So I went into the build_sdcard.sh file in the repo and saw that it is recommended to use a different version. It actually says to use 2020-08-24 but the link given is for 2021-04-09. The april one did not work for me but the august one did. 

This is a low priority issue but I wanted to point it out.

1) Broken image link (for me at least)
2) Inconsistent directions in the build_sdcard.sh file comments

openoms, 2022-01-17:
The hosting servers is being rebuilt in the meantime use the torrent: https://github.com/rootzoll/raspiblitz/raw/v1.7/home.admin/assets/raspiblitz-v1.7.1-2021-10-28.img.gz.torrent

The v1.7.1 is built on an older version of the raspios image: https://github.com/rootzoll/raspiblitz/blob/fd4256223c4ada1e1b4fdd8794792b1208514188/build_sdcard.sh#L5

We are working on a new SDcard release for which the latest release candidate can be downloaded from the `dev` branch.

GoetzGoerisch, 2022-01-18:
relates to #2899 and #2734

closed 2022-01-24 by rootzoll