raspiblitz/raspiblitz #2895: Connection issues persist after updating Tor from source

The34Goon, 2022-01-17:
Node is having trouble staying connected to peers and apps.
This stuff showed up out of the blue and didn't resolve when I updated Tor from source.

```
[NOTICE] Have tried resolving or connecting to address ...giving up [65 duplicates hidden]
[NOTICE] Tried for 120 seconds to get a connection to XYZ ..[15 duplicates hidden]
[WARN] connection_ap_attach_pending(): Bug: 0x55c8b31ce0 is no longer in circuit_wait. Its current state is waiting for rendezvous desc. Why is it on pending_entry_connections?
[NYX_NOTICE] Relay unresponsive
[WARN] Rejecting SOCKS request for anonymous connection to private address 127.0.0.1.
[NOTICE] Closed 2 streams for service XYZ.onion for reason resolve failed. Fetch status: No more HSDir available to query.
[WARN] Failed to launch rendezvous circuit to XYZ
```

Generating the debug link:
Unit blitzapi.service could not be found. (after a long wait it then exits without giving any link)
Tried again and the secound round was succesful:
[Debug output](https://termbin.com/u0sp)

Do you have any hints on how to keep my connections stable and resolve the problems?

openoms, 2022-01-17:
Sometimes I found a full node restart is needed to make these errors go.

The34Goon, 2022-01-17:
Did plenty of those...
Like just 30 minutes ago and now bitcoind is finding 1 peer and lnd 0/88

openoms, 2022-01-17:
Since I can't reproduce the issue it is hard to say anything.
Is Tor running ok on other machines on the same LAN?
Might worth it to restart the router too.

Also there is now a release candidate image in the `dev` you can update the whole system to if ok with testing.

The34Goon, 2022-01-17:
Yes, otherwise Tor works fine.  But there is always a glitch on the node after 2-3 days where I lose connection to peers and apps, RTL, Zeus.  It also triggers force closes here and there, which is probably normal when the htlc gets lost in lalaland.  I can't define it, but it feels like lnd and bitcoin don't communicate well with each other or sometimes bitcoin doesn't find any peers at all and therefore lnd doesn't start either.

Thx for the RC candidate I'll give it a try.

closed 2022-01-17 by The34Goon

The34Goon, 2022-01-20:
What is the best way to delete absolutely everything that is related to networking and tor but still keep my channels open and safe?

openoms, 2022-01-24:
For that would need to generate an lnd-rescue file (includes the whole lnd directory) and import it into a new installation.

To change the Tor address of the lnd node would need to delete the `v3_onion_private_key` in `/mnt/hdd/lnd` and restart lnd.