raspiblitz/raspiblitz #2896: SYSTEM Monitoring + Configuration

The34Goon, 2022-01-17:
When did the lightning options get removed from this menu?
[image]

openoms, 2022-01-17:
put
`lightning=on`
to your 
`/mnt/hdd/raspiblitz.conf`

closed 2022-01-17 by The34Goon

rootzoll, 2022-01-17:
@openoms `lightning` in conf should normally be `lnd` or `cl` (if those implementations installed & active)  ... the value `on` might make problems down the road.

The34Goon, 2022-01-17:
lnd is what I had in at first and then changed to =on, after which everything broke.

Now back to lnd but still no log option.

openoms, 2022-01-17:
sorry, was wrong from memory
so use:
`lightning=lnd`
and try adding:
`lnd=on`
if it is not there