raspiblitz/raspiblitz #2897: CL: test seed and expose the hsm-secret hex from menu

openoms, 2022-01-17:


openoms, 2022-01-27:
there is also https://github.com/rsbondi/clightning-go-plugin/tree/master/dump_keys to create an xpub and xpriv from the hsm_secret.

Will need to figure out how to build it and try.

openoms, 2022-02-02:
The next candidate is: https://github.com/jb55/clightning-dumpkeys
* needs `secp256k1`
```
cd ~/Downloads
git clone https://github.com/bitcoin-core/secp256k1.git
cd secp256k1
./autogen.sh
./configure
make
make check  # run the test suite
sudo make install
```
```
# configure dynamic links
sudo ldconfig
```

```
cd ~/Downloads
git clone https://github.com/jb55/clightning-dumpkeys
cd clightning-dumpkeys
make
```
Usage:
cd ~/Downloads/clightning-dumpkeys
sudo -u bitcoin ./clightning-dumpkeys /home/bitcoin/.lightning/bitcoin/hsm_secret
```

rootzoll, 2022-02-17:
@openoms can we close this issue or does this documentation still needs some transfere?

openoms, 2022-02-17:
will close this as recorded in the FAQ.cl.md, but will open a new one once I find a way to import a descriptor or xpub to an external wallet (other than the bitcoin-cli)

closed 2022-02-17 by openoms