raspiblitz/raspiblitz #2899: Build SD card method fails with "There is no tested version of python present"

nkinnan, 2022-01-18:
I am following the instructions here: https://github.com/rootzoll/raspiblitz#build-the-sd-card-image

Downloaded RaspiOS 64-bit here: https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/

Ran
```
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.7/build_sdcard.sh && sudo bash build_sdcard.sh
```

and got this output:
```
<snip>
E: Unable to locate package minecraft-pi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

*** Python DEFAULT libs & dependencies ***
!!! FAIL !!!
There is no tested version of python present
```

It's strange that something is looking for minecraft-pi but I assume the real error is the python one.  I am unsure how to proceed.  Is the RaspiOS 64-bit image I downloaded perhaps too new?  Is there a recommended image to use from raspberrypi.org?

nkinnan, 2022-01-18:
Figured out by reading build_sdcard.sh that the instructions at https://github.com/rootzoll/raspiblitz#build-the-sd-card-image are incorrect.  The version to use is not the latest, but https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-04-09/

Leaving this open since the documentation should be changed.

GoetzGoerisch, 2022-01-18:
This incorrectness is due to the switch from `buster` to `bullseye`.
As you found out, the actual tested image incl. `sha256sum` is listed in the header of `build_sdcard.sh`.

Already fixed by #2734 for `dev` and will be merged for `1.7.2`.

ping @rootzoll

nkinnan, 2022-01-18:
Thanks.  It would be nice to correct the instructions on the main page in the meantime to avoid confusion.

closed 2022-01-24 by rootzoll

konigra, 2022-02-05:
Hello, 

I ran into the same error unforunately.

Not a smooth "do it yourself updating experience".

But thats part of the job, right? 
Need to flash the SD card again then. :)

[image]

rootzoll, 2022-02-06:
A v1.7.2 with updated build script is in the works ... if you like you can try to build with `dev` branch and give feedback on that: `wget --no-cache https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh && sudo bash build_sdcard.sh -b dev`

konigra, 2022-02-06:
I solved it by using not the latest raspberry OS version and I followed the "left" n00bie path.
[image]

Then I had to "compact" the channel.db, because the former one was too big.
It was 77G (!) compared to now 140M.

[image]

Where do you want me to provide feedback on that? Right here?
I would love to help out with some issues!!

rootzoll, 2022-02-06:
Version 1.7.2 (see `dev` branch) has already a compact feature prepared - if you have further feedback on db compaction you can post it here: https://github.com/rootzoll/raspiblitz/issues/2752

There will be a Release Candidate #2 of v1.7.2 within the next days - that can be build or downloaded from the dev branch. If you like to help out, best would be to test that Release Candidate and see what issues are still open for you.